greatscottgadgets/ubertooth #149: ubertooth-rx segfaults on first call to cb_rx

dominicgs, 2016-02-02:
```
dominicgs@hydrogen:139:~/CS/Bluetooth/ubertooth/host/build$ gdb ubertooth-rx 
GNU gdb (Ubuntu 7.10-1ubuntu2) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ubertooth-rx...done.
(gdb) r
Starting program: /usr/local/bin/ubertooth-rx 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6cec700 (LWP 32719)]

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7726415 in btbb_piconet_get_lap (pn=0x0) at /home/dominicgs/CS/Bluetooth/libbtbb/lib/src/bluetooth_piconet.c:103
103		return pn->LAP;
(gdb) bt
#0  0x00007ffff7726415 in btbb_piconet_get_lap (pn=0x0) at /home/dominicgs/CS/Bluetooth/libbtbb/lib/src/bluetooth_piconet.c:103
#1  0x0000000000401ac2 in cb_rx (ut=0x604010, args=0x0) at /home/dominicgs/CS/Bluetooth/ubertooth/host/ubertooth-tools/src/ubertooth-rx.c:88
#2  0x00007ffff7bd10ca in ubertooth_bulk_receive (ut=0x604010, cb=0x401980 <cb_rx>, cb_args=0x0) at /home/dominicgs/CS/Bluetooth/ubertooth/host/libubertooth/src/ubertooth.c:274
#3  0x000000000040226c in main (argc=1, argv=0x7fffffffde38) at /home/dominicgs/CS/Bluetooth/ubertooth/host/ubertooth-tools/src/ubertooth-rx.c:281
(gdb)
```

dominicgs, 2016-02-02:
@elli89 Does this look like it's related to your changes?

elli89, 2016-02-02:
Yes. I swapped two code lines so the piconet wasn't allocated.

I have found and fixed another bug made by me in the callback function. In my changes I assumed that the uap and lap of a piconet would automatically be initialized with UAP_ANY and LAP_ANY. So I didn't tell the callback method to search for any UAP if none was provided.

I will submit a fix in a minute.

closed 2016-02-02 by dominicgs