greatscottgadgets/ubertooth #155: ubertooth-scan add flag to disable name resolution

ZeroChaos-, 2016-02-10:
right now, when ubertooth finds a UAP it does some kind of remote name request using the bluetooth card.  I'd like to be able to stop it from doing that.  a simple --no-name or whatever would be great.

Example:
??:??:3F:A9:BC:CD       Pwnie

What I want is for it to not touch my bluetooth card and return either:
??:??:3F:A9:BC:CD       [Unknown]         <- this his what it shows now when it fails to get a remote name
or
??:??:3F:A9:BC:CD

dominicgs, 2016-02-10:
This is now supported as part of ubertooth-rx.  Survey mode can be activated using `ubertooth-rx -z`, the timeout can be adjusted using `ubertooth-rx -z -t <timeout>`

maximcherny, 2016-02-26:
Thanks for this feature! However the summary is not being printed when a script running with -t 0 is terminated via CTRL+C. Should this be the case?

elli89, 2016-02-26:
@maximcherny thanks for the hint. You're right, it makes sense to display the scan results when the program is terminated. I have written a patch for libubertooth.

dominicgs, 2016-02-26:
I have merged #160, thank @elli89 

I'll close this now as I believe that this feature is now working as expected.

closed 2016-02-26 by dominicgs