greatscottgadgets/ubertooth #157: Kismet plugins do not handle mutlipel connected Ubertooth devices

dominicgs, 2016-02-19:
From the mailing list:

```
I have two Ubertooth Ones which I want one to use as spectrum analyzer and the second one
as a sniffer within kismet at the same time. I know that I can specify the ubertooth device
which to used by using -U. But inside kismet I am not able to choose between my ubertooth
devices no matter what I type (e.g. ubertooth, ubertooth -U0, ubertooth -u0, ubertooth-U0,
ubertooth-u0, ubertooth:U0 …) kismet will show the following message inside the console

„multiple Ubertooth devices found! Use '-U' to specify device number
Device 0: Serial No: xxxx
Device 1: Serial No: yyyy"

When I am using only one ubertooth device kismet is working fine. I can plug in my
second ubertooth and let it run as an spectrum analyzer.
```

dominicgs, 2016-02-20:
Two options: allow this to be specified in the config file or just have Kismet use the first Ubertooth.

I'll probably only do the second one in time for the March release.

elli89, 2016-03-16:
The kismet plugin already has an option in the source code to choose one a specific Ubertooth, but it is commented. I uncommented and tested it and was able to run kismet using two Ubertooth sticks simultaneously. Is there a reason why the code is commented? If not I could file a PR.

dominicgs, 2016-03-16:
I have no idea why it was commented out.  Perhaps I was hoping to be able to specify which Ubertooth is used?  Or does it already do that?

elli89, 2016-03-16:
With the lines uncommented and some minor fixes it is already possible to tell kismet which Ubertooth device to use.