greatscottgadgets/ubertooth #159: libUSB Error: Command Error:  (4)

maximcherny, 2016-02-26:
When running the latest libbtb and ubertooth (at the time of writing), I am getting:

```
ubertooth-rx -s -z -t 0
```

```
offset > CLK_TUNE_TIME
CLK100ns Trim: 520
libUSB Error: Command Error:  (4)
```

```
Linux [redacted] 3.19.0-25-generic #26~14.04.1-Ubuntu SMP Fri Jul 24 21:18:00 UTC 2015 i686 i686 i686 GNU/Linux
```

Then the command proceeds to run normally. Is this an indication of any issue or should it be ignored in the meantime? Was not present in the latest stable build.

elli89, 2016-02-26:
Try updating the ubertooth firmware to the latest git version. It looks like your currently used firmware doesn't understand the clock trimming USB command yet (it is quite new).

maximcherny, 2016-03-01:
Thanks for the tip, that fixed it!

dominicgs, 2016-03-03:
Firmware / host code mismatch.  Fixed by using matched versions.

closed 2016-03-03 by dominicgs