greatscottgadgets/ubertooth #161: Writting my own code

MrSarinth, 2016-02-27:
Hi, is there a way to write my own code in JAVA or C or anything and make Ubertooth just do stuff? Like listen on specific frequency and print recieved data. Or transmit my own data packates.

I am looking for any documentation or guide to help me write something and show what you can do with Ubertooth. Or perhaps just anything to help me understand how to achieve any of this and I could write that guide myself. 

Thanks

dominicgs, 2016-02-28:
There is a C library (libUbertooth) to control the Ubertooth.  It handles all of the functions of the command line tools, such as tuning to a frequency and returning data which can then be processed to find Bluetooth packets or used for anything else you may want to find.

As the interface to the library has changed a lot over the years, it is not currently well documented.

Is there something that you would like to achieve with Ubertooth that is not currently possible with the tools provided?

MrSarinth, 2016-02-28:
I am more interested what I can do with Ubertooth.

dominicgs, 2016-02-28:
I'm not sure what you mean by that.  Are you saying that you're interested in the capabilities of the hardware, rather than the features of the software?

MrSarinth, 2016-02-28:
I suppose I am no sure either. I am supposed to figure out what I can do with it. Besides things in section Getting Started. I would like to write simple code or use command line to transmit signal or listen to specific frequency. Basic stuff really. I think that library is all I need. I wanted to know if there is anything else or anything more beginner friendly.

closed 2020-12-03 by mikeryan