greatscottgadgets/ubertooth #162: kismet_ubertooth build failure in git

ZeroChaos-, 2016-02-28:
```
>>> Jobs: 0 of 1 complete, 1 failed                 Load avg: 0.58, 2.01, 1.97
 * Package:    net-wireless/kismet-ubertooth-2015.10.1
 * Repository: maintainer-zero
 * Maintainer: zerochaos@gentoo.org
 * USE:        abi_x86_64 amd64 elibc_glibc kernel_linux userland_GNU
 * FEATURES:   preserve-libs sandbox userpriv usersandbox
>>> Unpacking source...
>>> Unpacking ubertooth-2015-10-R1.tar.xz to /var/tmp/portage/net-wireless/kismet-ubertooth-2015.10.1/work
>>> Source unpacked in /var/tmp/portage/net-wireless/kismet-ubertooth-2015.10.1/work
>>> Preparing source in /var/tmp/portage/net-wireless/kismet-ubertooth-2015.10.1/work/ubertooth-2015-10-R1 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/kismet-ubertooth-2015.10.1/work/ubertooth-2015-10-R1 ...
>>> Source configured.
>>> Compiling source in /var/tmp/portage/net-wireless/kismet-ubertooth-2015.10.1/work/ubertooth-2015-10-R1 ...
make -j1 KIS_SRC_DIR=/usr/include/kismet/ 
x86_64-pc-linux-gnu-g++ -I/usr/include/ncurses -I/usr/include/libnl3  -Wall -Os -mtune=nocona -pipe -frecord-gcc-switches -march=native -I/usr/include/kismet/ -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o 
packetsource_ubertooth.cc: In member function ‘virtual int PacketSource_Ubertooth::OpenSource()’:
packetsource_ubertooth.cc:267:12: error: cannot convert ‘ubertooth_t*’ to ‘libusb_device_handle*’ in assignment
  if ((devh = ubertooth_start(-1)) == NULL) {
            ^
packetsource_ubertooth.cc:280:22: error: cannot convert ‘libusb_device_handle*’ to ‘ubertooth_t*’ for argument ‘1’ to ‘void ubertooth_stop(ubertooth_t*)’
   ubertooth_stop(devh);
                      ^
packetsource_ubertooth.cc:288:22: error: cannot convert ‘libusb_device_handle*’ to ‘ubertooth_t*’ for argument ‘1’ to ‘void ubertooth_stop(ubertooth_t*)’
   ubertooth_stop(devh);
                      ^
packetsource_ubertooth.cc: In member function ‘virtual int PacketSource_Ubertooth::CloseSource()’:
packetsource_ubertooth.cc:317:22: error: cannot convert ‘libusb_device_handle*’ to ‘ubertooth_t*’ for argument ‘1’ to ‘void ubertooth_stop(ubertooth_t*)’
   ubertooth_stop(devh);
                      ^
Makefile:54: recipe for target 'packetsource_ubertooth.o' failed
make: *** [packetsource_ubertooth.o] Error 1
 * ERROR: net-wireless/kismet-ubertooth-2015.10.1::maintainer-zero failed (compile phase):
 *   emake failed

dominicgs, 2016-02-29:
I can't reproduce this error.  It looks like you may be building the release plugins against the git library.  Is that possible?

ZeroChaos-, 2016-03-04:
yes, that's exactly what happened. -EINVALID

closed 2016-03-04 by ZeroChaos-