greatscottgadgets/ubertooth #163: Upgrading the Firmware on Ubertooth One

sumeetjauhar, 2016-03-07:
I have been trying to the the ubertooth one started on my ubuntu virtual machine but i have run into some problems. 

I followed the installation procedure and now i can see the bluetooth spectrum via the ubertooth-specan-ui command. 

However i believe that the firmware on my board is incorrect 
ubertooth-util -v
control message unsupported
Firmware revision: error: -9

However attempting to update the firmware seems a tad bit annoying as i get the same "control message unsupported" error. 

I believe i hv the same problem as discussed here https://sourceforge.net/p/ubertooth/mailman/message/34582005/ . The fix for it seems to be to jumper the board and force it into DFU mode. 

Now i can see the LED flashing for 5 seconds app. But hey i am running a VM and i need to take the USB back into the VM and then run the update firmaware command all in `5 seconds` !! 

Is there no easy solution to this problem ? Or do i need a base linux operating system (which i dont have )

sumeetjauhar, 2016-03-07:
Also i automount the Ubertooth to be auto detected on my guest Ubuntu but honestly it takes more than 5 seconds  to get attached .

While it is not yet detected I get the following error : 
ubertooth-dfu -d ubertooth-2015-10-R1/ubertooth-one-firmware-bin/bluetooth_rxtx.dfu -r 
could not open Ubertooth device
Segmentation fault (core dumped)

When it does get detected ( after 5+sec) I get the a long string of this error message : 
control message unsupported
control message unsupported
control message unsupported

Any inputs on how to resolve this problem ?

dominicgs, 2016-03-07:
I'm honestly surprised that it takes more than 5 seconds to get attached to the VM, I've never heard of it taking more than a second or two.

Do you have both USB IDs set up for automatic pass through to the VM?  The regular firmware ID and the DFU ID?
0x1d50 0x6002  Ubertooth One
0x1d50 0x6003  Ubertooth One DFU

If you don't have the DFU ID passed through, it won't appear until after the timeout and the switch back to regular mode.

sumeetjauhar, 2016-03-07:
How do i check that both the USB id have passed . I can see only one 

osboxes@osboxes:~$ lsusb 
Bus 001 Device 013: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 015: ID ffff:0004  
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

sumeetjauhar, 2016-03-07:
[image]

dominicgs, 2016-03-07:
Ah, you have a slightly older model device where the DFU ID is 0xFFFF 0x0004.  You need to create VirtualBox "USB Device Filters" for each of the IDs.

sumeetjauhar, 2016-03-08:
Yes correct, I bought the Ubertooth hardware quite some time ago but didnt get around to use it.

I dont see the other ID ( Assuming you mean the DFU ID here) . I showed you the screen shot of the Virtual Box USB settings above. How do i enumerate the other ID and add the filter for it ?

sumeetjauhar, 2016-03-08:
Alright a bit more trial and error and finally got the Virtual box to detect the USB within 5 seconds and was able to re-flash my firmware . 

osboxes@osboxes:~$ ubertooth-util -v
Firmware revision: 2015-10-R1
osboxes@osboxes:~$ ubertooth-util -V
ubertooth 2015-10-R1 (dominicgs@hydrogen) Mon Oct 12 22:30:56 BST 2015
osboxes@osboxes:~$ 

Seems good for the moment. Time to setup Kismet and Wireshark.

Thanks for the support.

dominicgs, 2016-03-08:
That's great, I'm glad you managed to get that working.

closed 2016-03-08 by dominicgs

12424642, 2016-08-17:
Hi Sumeetjauhar and dominicgs, I'm currently facing the exact problem. I've tried making changes to the filter but can seem to resolve it. 

Every time I try to update the firmware I get 'could not open Ubertooth device' followed by 'Segmentation fault (core dumped)'

Pleaseee help, 
Thanks in advance!

12424642, 2016-08-17:
*can't

dominicgs, 2016-08-17:
Could you give some information about the problem you are having.  The USB IDs that you have filtered?  The IDs that you see from the VM?  The IDs that you see on the host?

12424642, 2016-08-17:
When I attempt to run the following command ‘ubertooth-dfu -d bluetooth_rxtx.dfu -r’ I receive ‘could not open Ubertooth device' followed by 'Segmentation fault (core dumped)'.
Another time it began briefly and the chasing lights lit up on the ubertooth device, but for some reason it gave the following message ‘……………………………………………………………. detached’, the chasing lights continued till I ran the command again.

12424642, 2016-08-17:
[image]

12424642, 2016-08-17:
[image]

12424642, 2016-08-17:
[image]

12424642, 2016-08-17:
These are the filters I set up and the IDs on both the main system as well the vm.

dominicgs, 2016-08-17:
Do you also have a filter set for the Ubertooth when it is in update mode?  It has a different USB ID then - either 0xffff 0x0004, or 0x1d50 0x6003

12424642, 2016-08-17:
So I should add another filter which uses 0xffff 0x0004 so that ubertooth can be identified while updating? 
You are possible right because I tried doing all the steps again on my kali machine and it gave me 'switching to DFU mode ...' and 'Unable to find Ubertooth'

dominicgs, 2016-08-17:
Yes, add the additional filter and let me know if it works.

12424642, 2016-08-17:
I've tried it and i'm still getting the same error initially, but when I try running the command again directly afterwards, it gives me the message 'checking firmware signature.....................................Detached'

12424642, 2016-08-17:
If I run the spectrum tool and that detects the Ubertooth, does that confirm the Ubertooth is at least working?

12424642, 2016-08-17:
It's very strange, I used the command 'ubertooth-btle –p -c capture.pcap' and I am able to see the ubertooth device working on the terminal. Still can't seem to carry out a firmware check or update. 
Sorry this might be a bit irrelevant but is there command is used to stop the capture and save it to pcap? I seem to just have to close the terminal.

dominicgs, 2016-08-17:
`checking firmware signature.....................................Detached` isn't an error message.  It's a successful firmware load.

12424642, 2016-08-17:
Oh sorry I didn't realise, thank you so much for your help!

ubertooth, 2017-06-19:
We have installed the ubertooth in ubuntu 14.04.

root@zan:~/ubertooth-2017-03-R2/firmware/bluetooth_rxtx# ubertooth-util -v
Firmware version: 2017-03-R2 (API:1.02)

root@zan:~/ubertooth-2017-03-R2/firmware/bluetooth_rxtx# ubertooth-util -V
ubertooth 2017-03-R2 (root@zan) Fri Jun 16 17:43:55 IST 2017

root@zan:~/ubertooth-2017-03-R2/firmware/bluetooth_rxtx# ubertooth-dfu -d bluetooth_rxtx.dfu -r
Switching to DFU mode...
Checking firmware signature
........................................
........................................
.................
Detached

I have got the above msg in console.

Please help

dominicgs, 2017-06-25:
2017-03-R2 is the latest firmware.  The output from `ubertooth-dfu` is as expected.