greatscottgadgets/ubertooth #167: Ubertooth-scan, Finding non-discoverable devices

garyv100, 2016-03-14:
Hi Dominic, 

I am trying to replicate your YouTube demo: HCI scan, Finding non-discoverable devices.

This is what I do and get:
/****************************************
sudo ./ubertooth-scan -s -x
HCI scan

Ubertooth scan

Scan results:
/****************************************
sudo ubertooth-scan -b hci0 -t 40 -s -x 
HCI scan

Ubertooth scan

Scan results:
/****************************************

I have just updated to your latest release as of 03/12/2016, rebuild and reloaded the firmware.
 ./ubertooth-util -V
ubertooth git- (garyv1@tech) Mon Mar 14 13:11:40 EDT 2016

./ubertooth-util -v
Firmware revision: git- 

ubertooth-rx -V
libubertooth 0.3 (GITDIR-NOTFOUND), libbtbb 0.4 (GITDIR-NOTFOUND)

I had the same issue w/ previous release: 2015-10-R1.

My environment: ubuntu 14.04 LTS on metal (no vm of any kind). I use Trandnet bluetooth along with Ubertooth One. I can successfully detect MAC of a bluetooth devices using: "hcitool inq".

Please correct me if i'm wrong, ubertooth-scan -s -x retrieves MAC of nearby bluetooth devices (i have a mix of Android and iPhone devices in one foot radius) in a non-discoverable mode or even if bluetooth is disabled - as long as they are powered? 

What am I doing wrong?

Best regards,

Gary

dominicgs, 2016-03-15:
I'm not sure what you're doing wrong, it looks like your Ubertooth isn't seeing any data.

As you are running the git code, could you try using `ubertooth-rx -z` ?

dominicgs, 2016-03-15:
Let me rephrase that - I don't think you're doing anything wrong, I think your Ubertooth is just not seeing any packets.

garyv100, 2016-03-15:
Dominic, you are correct, the bt was not sending any data and "ubertooth-rx -z" returns no data.
If I, for example, transfer a file over bt - it does return data and I get the following:
/******************************
./ubertooth-scan -s -x
HCI scan

Ubertooth scan
systime=1458054711 ch= 2 LAP=3d6bb7 err=0 clk100ns=1303146539 clk1=1257079 s=-36 n=-74 snr=38
systime=1458054712 ch=66 LAP=3d6bb7 err=1 clk100ns=1303997799 clk1=1257215 s=-53 n=-73 snr=20
systime=1458054712 ch=19 LAP=3d6bb7 err=0 clk100ns=1304371648 clk1=1257275 s=-16 n=-67 snr=51

Scan results:
??:??:85:3D:6B:B7	LGMS323
Requesting information ...
Can't create connection: Operation not permitted
AFH map: 0x048f5808001034055c53
/*********************************

Please do correct me if i'm wrong. In order for Ubertooth One to retrieve MAC it must receive data packets from that device and in doing so it reconstructs MAC?  If it does not get data, it cannot do it?

Many thanks,

Gary

PS: Love your device, a lot of hard work and dedication went in to it.  In my previous life i used to do similar things - I can appreciate it.

dominicgs, 2016-03-15:
That's correct, Ubertooth is sniffing packets and reconstructing the lower four bytes of the MAC.  You can then take those bytes and try to connect to the device (putting 00:00 for the first two bytes).

As you can see from the 'HCI scan' portion of your output, the device was not discoverable to the Bluetooth dongle.

garyv100, 2016-03-15:
Dominic, thank you so much!

closed 2016-03-15 by garyv100

bigoper, 2023-11-02:
Hi @garyv100 and @dominicgs 

I was hoping I could use your help here :)

# CURRENT STATE
### `ubertooth-scan -s -x`
- No BT device is being actively used used/sync/advertising
- I have laptops, phones, iPads near me with BT activated/on
```js
└─$ ubertooth-scan -s -x
HCI scan

Ubertooth scan
systime=1698884138 ch=26 LAP=07e88f err=2 clk100ns=26085068 clk1=4174 s=-80 n=-55 snr=-25
systime=1698884146 ch=66 LAP=a383e7 err=2 clk100ns=97813755 clk1=15650 s=-81 n=-55 snr=-26

Scan results:
??:??:??:07:E8:8F
AFH map: 0x00000004000000000000
??:??:??:A3:83:E7
AFH map: 0x00000000000000000400
```
```js
└─$ ubertooth-scan -s -x
HCI scan

Ubertooth scan
systime=1698884387 ch=72 LAP=a9dfd5 err=1 clk100ns=40549063 clk1=6488 s=-75 n=-55 snr=-20

Scan results:
??:??:??:A9:DF:D5
AFH map: 0x00000000000000000001
```
- You can see that the results are NOT consistent.

### `ubertooth-rx -z`
- Same devices as before, same state.
```js
└─$ ubertooth-rx -z     
systime=1698884436 ch=74 LAP=a9dfd5 err=0 clkn=33837 clk_offset=4491 s=-74 n=-55 snr=-19
systime=1698884437 ch=67 LAP=a9dfd5 err=1 clkn=36653 clk_offset=4354 s=-76 n=-55 snr=-21
systime=1698884440 ch=69 LAP=a9dfd5 err=2 clkn=47397 clk_offset=4160 s=-74 n=-55 snr=-19
systime=1698884451 ch=68 LAP=a9dfd5 err=2 clkn=82533 clk_offset=3654 s=-75 n=-55 snr=-20
systime=1698884460 ch= 2 LAP=a9dfd5 err=2 clkn=111913 clk_offset=3454 s=-73 n=-55 snr=-18
systime=1698884463 ch=66 LAP=a9dfd5 err=2 clkn=122313 clk_offset=3281 s=-74 n=-55 snr=-19
systime=1698884479 ch= 4 LAP=0c3711 err=1 clkn=173297 clk_offset=4127 s=-77 n=-55 snr=-22
systime=1698884491 ch=77 LAP=a9dfd5 err=2 clkn=211712 clk_offset=2256 s=-72 n=-55 snr=-17
systime=1698884524 ch=72 LAP=a9dfd5 err=1 clkn=316376 clk_offset=909 s=-78 n=-55 snr=-23
systime=1698884540 ch=70 LAP=465b37 err=0 clkn=366254 clk_offset=2489 s=-63 n=-55 snr=-8
systime=1698884540 ch=70 LAP=465b37 err=0 clkn=366256 clk_offset=2474 s=-54 n=-55 snr=1
systime=1698884541 ch=69 LAP=a9dfd5 err=1 clkn=371028 clk_offset=125 s=-75 n=-55 snr=-20
systime=1698884561 ch=60 LAP=a9dfd5 err=0 clkn=433799 clk_offset=5218 s=-74 n=-55 snr=-19
systime=1698884571 ch=66 LAP=a9dfd5 err=1 clkn=466155 clk_offset=4829 s=-74 n=-55 snr=-19
systime=1698884586 ch=77 LAP=a9dfd5 err=1 clkn=515107 clk_offset=4120 s=-73 n=-55 snr=-18
systime=1698884619 ch= 8 LAP=c360de err=2 clkn=619480 clk_offset=2767 s=-82 n=-55 snr=-27
systime=1698884620 ch=50 LAP=07eae6 err=2 clkn=622821 clk_offset=4551 s=-75 n=-55 snr=-20
systime=1698884652 ch=60 LAP=07eae6 err=2 clkn=727132 clk_offset=314 s=-78 n=-55 snr=-23
systime=1698884656 ch=68 LAP=a9dfd5 err=2 clkn=739818 clk_offset=956 s=-74 n=-55 snr=-19
^CSurvey Results
??:??:??:A9:DF:D5
??:??:??:0C:37:11
??:??:??:46:5B:37
??:??:??:C3:60:DE
??:??:??:07:EA:E6
```
- I'm detecting more devices.

# NEW STATE
- Same devices, but now they'll be in discoverable mode.
- I'm doing this^ so I can compare the results to make sure that LAP that I'm seeing are my devices.
[... 108 lines cut ...]
systime=1698885619 ch= 0 LAP=5e749b err=0 clkn=70816 clk_offset=806 s=-51 n=-55 snr=4
systime=1698885619 ch=64 LAP=465b37 err=0 clkn=71126 clk_offset=2262 s=-54 n=-55 snr=1
systime=1698885619 ch=64 LAP=465b37 err=0 clkn=71128 clk_offset=2251 s=-54 n=-55 snr=1
systime=1698885619 ch=49 LAP=465b37 err=2 clkn=71366 clk_offset=2262 s=-54 n=-55 snr=1
systime=1698885619 ch=34 LAP=465b37 err=0 clkn=71630 clk_offset=2240 s=-58 n=-55 snr=-3
systime=1698885619 ch=34 LAP=465b37 err=0 clkn=71632 clk_offset=2245 s=-53 n=-55 snr=2
systime=1698885620 ch=68 LAP=465b37 err=0 clkn=72392 clk_offset=2234 s=-56 n=-55 snr=-1
systime=1698885620 ch=55 LAP=465b37 err=0 clkn=73256 clk_offset=2234 s=-53 n=-55 snr=2
systime=1698885620 ch=40 LAP=465b37 err=0 clkn=73582 clk_offset=2223 s=-54 n=-55 snr=1
systime=1698885620 ch=40 LAP=465b37 err=0 clkn=73584 clk_offset=2218 s=-54 n=-55 snr=1
systime=1698885620 ch=72 LAP=240d6e err=2 clkn=73726 clk_offset=276 s=-87 n=-55 snr=-32
systime=1698885621 ch=65 LAP=465b37 err=0 clkn=76498 clk_offset=2238 s=-53 n=-55 snr=2
systime=1698885621 ch=52 LAP=465b37 err=0 clkn=77390 clk_offset=2207 s=-53 n=-55 snr=2
systime=1698885621 ch=52 LAP=465b37 err=1 clkn=77392 clk_offset=2198 s=-53 n=-55 snr=2
^CSurvey Results
??:??:??:35:CC:1B
??:??:4F:46:5B:37
??:??:C2:59:A7:C4
??:??:88:5E:74:9B
??:??:??:24:0D:6E
```
- My Apple devices are **not** in the list of devices found.
Not too sure how this thing works/operates, as the results are confusing.

Thanks for looking!