greatscottgadgets/ubertooth #171: Bluetooth BLE Capture errors

Nocks932, 2016-03-21:
Hello,
I managed to setup my Ubertoothone using the guild and also tried to update the firmware on it. Im currently running an Ubuntu 14.04LTS, dual boot with win10, so it's not a virtual machine.
When I try to run the : ubertooth-btle -f -c /tmp/pipe i get a strange message : control message unsupported, with no error code to it.
The current firmware of the device is : Firmware revision: 2014-04-R1
Can you help me with this issue ?

dominicgs, 2016-03-29:
@Nocks932 Which version of the host tools are you using?  This error is normally a symptom of a host/firmware version mismatch.  I think you probably need to upgrade your firmware to release 2015-10-R1 to match the host code.

Nocks932, 2016-03-29:
Hey Dominic, 
I tried updating the firmware but it doesnt find the ubertooth dongle. The lights are flashing as they should but it doesnt recognise it. I'm currently running ubuntu LTS 14 if that helps.
Thanks in advance.

dominicgs, 2016-03-29:
When the lights are flashing, what shows up when you run `lsusb` ?

Nocks932, 2016-03-29:
Bus 004 Device 004: ID 8086:0189 Intel Corp. 
Bus 004 Device 003: ID 04d9:a067 Holtek Semiconductor, Inc. 
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 1bcf:2881 Sunplus Innovation Technology Inc. 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
This is what i get when i run the command

dominicgs, 2016-03-30:
That's very strange, when the lights are flashing it should enumerate with a USB ID of either FFFF:0004 or 1d50:6003

If you connect pins 1 and 3 on the [expansion header](https://github.com/greatscottgadgets/ubertooth/wiki/Ubertooth-One#pins-and-leds) while plugging it in, it should enter DFU mode too.  Could you try that and then `'lsusb` (beware, it has a short timeout before loading the normal firmware)

Nocks932, 2016-03-30:
Hello Dominic,
I tried what you said and this is what happened. I connected pins 1 and 3 as you said and attached it to my computer and the lights started flashing ( i gues this means its entering DFU mode ? ) i immediately typed the lsusb command as you asked and the only different entry i get is : 
Bus 001 Device 012: ID 1d50:6002 OpenMoko, Inc. 
I'm assuming this is it because the other entries in the list are the same.
Hope this helps.

Nocks932, 2016-04-03:
Hello again DOminic,
Were you able to identify this issue because i need the device working as soon as possible.
Thanks in advance,
Martin.

dominicgs, 2016-04-05:
@Nocks932 The issue appears to be that your host does not see Ubertooth in DFU mode, `Bus 001 Device 012: ID 1d50:6002 OpenMoko, Inc.` is the Ubertooth in regular firmware mode.  This may be a timing thing, the lights flashing suggest that it gets in to DFU mode but it is timing out and entering the regular firmware before that happens.

Could you try the following:
 * Insert the Ubertooth and confirm that `ID 1d50:6002 OpenMoko, Inc.` appears in lsusb
 * Run `ubertooth-util -f`
 * Run `lsusb` again and check for a different device?

closed 2018-06-27 by mikeryan