greatscottgadgets/ubertooth #173: Improve handling of firmware version mismatch

mikeryan, 2016-03-25:
@richo suggests that our message be something more along the lines of "new firmware is available!"

At a minimum we should include a link to a Wiki page about updating firmware, optimizing for the common case of someone using a tarball.

In the medium/long term I'd like to include auto-update capabilities that can fetch firmware bins from somewhere. My perceived workflow:

- user tries to run ubertooth-something, gets error message about new firmware
- user runs ubertooth-dfu --autoupdate
- autoupdate fetches firmware from (?), maybe ubertooth.github.io
- signature verified against @dominicgs's GPG key
- burn baby burn

Other concerns: we should make ubertooth.github.io redirect somewhere sensible, either the GH project page or greatscottgadgets.com. What do we do if Dom gets hit by a bus? Presumably anyone with a commit bit can update the repo and add a new GPG fingerprint.

More wishlist: check the userspace code version and link users to the latest release if they're out of date.

dominicgs, 2016-03-29:
I'll probably split this in to two parts:
1) Making the error message more friendly / helpful
2) Everything else

EMCP, 2016-05-16:
speaking of which, I just ran the following and got a bit scared that the firmware isn't updated.. but it seems to be reporting the latest version..?

```
root@kali:~/Tools/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-dfu -d bluetooth_rxtx.dfu -r
Switching to DFU mode...
Checking firmware signature
.......................................................................................................................................
control message unsupported
root@kali:~/Tools/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-
ubertooth-btle       ubertooth-dump       ubertooth-rx         ubertooth-specan-ui
ubertooth-debug      ubertooth-ego        ubertooth-scan       ubertooth-util
ubertooth-dfu        ubertooth-follow     ubertooth-specan     
root@kali:~/Tools/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-
ubertooth-btle       ubertooth-dump       ubertooth-rx         ubertooth-specan-ui
ubertooth-debug      ubertooth-ego        ubertooth-scan       ubertooth-util
ubertooth-dfu        ubertooth-follow     ubertooth-specan     
root@kali:~/Tools/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-util -v
Firmware revision: 2015-10-R1

```

EMCP, 2016-05-16:
Wiki seems to mention a detach command, but that appears to be no longer available?

"Press enter, and the device will automatically enter DFU mode and flash the firmware. When done, you can return it into regular operation mode by unplugging and replugging it, or running `ubertooth-dfu --detach.`"

```
root@kali:~/Tools/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-dfu --detach
ubertooth-dfu: invalid option -- '-'
ubertooth-dfu - Ubertooth firmware update tool
Usage:
	-h this help
	-s <filename> add DFU suffix to binary firmware file
	-u <filename> upload - read firmware from device
	-d <filename> download - write DFU file to device
	-r reset Ubertooth after other operations complete
	-U <0-7> set ubertooth device to use
root@kali:~/Tools/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-dfu --detach

```

dominicgs, 2016-05-16:
@EMCP That's expected output and unrelated to this issue.  It it the Ubertooth resetting itself that causes the `control message unsupported` warning, but that's what we want it to do.  One day I will find a way to sensibly suppress that message.