greatscottgadgets/ubertooth #174: why i upgrade  hint:libUSB Error: Timeout:  (-7)

Robinchenjiaquan, 2016-03-29:
root@ubuntu:/home/robin/Downloads/ubertooth-2015-09-R2/ubertooth-one-firmware-bin# ubertooth-dfu -d bluetooth_rxtx.dfu -rlibUSB Error: Timeout:  (-7)
Switching to DFU mode...
Unable to find Ubertooth
root@ubuntu:/home/robin/Downloads/ubertooth-2015-09-R2/ubertooth-one-firmware-bin#

linuxn00b, 2016-03-29:
Try the following command to dump more:
LIBUSB_DEBUG=8 ubertooth-dfu -d bluetooth_rxtx.dfu -r

You may find what's going on from the debug info.

dominicgs, 2016-03-29:
@Robinchenjiaquan are you running in a virtual machine?  Ubertooth changes USB ID when it switches to DFU mode, if you're passing it through to the VM, then you'll need to pass through the DFU ID too.

closed 2016-05-24 by dominicgs

luxuguang-leo, 2019-03-12:
> @Robinchenjiaquan are you running in a virtual machine? Ubertooth changes USB ID when it switches to DFU mode, if you're passing it through to the VM, then you'll need to pass through the DFU ID too.

Could you tell me how to do that in details?I got this issue in VM, but seems every similar have no final soluction. Thanks