greatscottgadgets/ubertooth #175: determining master's clock for hopping calculation

michaelp123, 2016-04-03:
Does ubertooth determine the master's clock that determines the hopping sequence?
If yes, how does it do it?

dominicgs, 2016-04-03:
Yes, there's a blog post on it here: http://ubertooth.blogspot.co.uk/2014/06/discovering-bluetooth-uap.html

michaelp123, 2016-04-04:
Great! 
But this only shows the 6 lower bits of the clock. How can I know the whole 28 bits of the clock (for calculating hopping sequence)?

dominicgs, 2016-04-04:
I'm sorry, I thought that post went on to explain the rest.

Once we have the UAP and LAP we calculate the entire hopping pattern (all 2^27 timeslots).  Then we take the observed packets, their channels, and their timestamps, and we try to fit them in to the hopping pattern.  In general it takes around six observed packets to find only one point in the hopping pattern where the set of packets fit.  This gives us the clock value.

Does that explain things?  Is there any part that you would like clarified?

michaelp123, 2016-04-04:
I don't completely understand. Once you have LAP and UAP and CLK1-6, you need all remaining 21 bits of the clock to determine the hopping sequence. Are you doing any brute force actions? Can you specify the specific actions (like a pseudo algorithm) you do after you have all these 3 parameters? (LAP, UAP and CLK1-6)

dominicgs, 2016-04-04:
Yes, it's a form of brute force.  We calculate 2^27 hop values, i.e. for the entire range of clock values.

michaelp123, 2016-04-04:
OK let's assume you guessed a specific clock value. What do you do now? calculate the hopping sequence? how do you know you made the right selections? (of clock and hop sequence)

dominicgs, 2016-04-04:
We aren't guessing anything.  The channel is a function of LAP, UAP and CLK.  The hopping pattern is the progression of channels over time.  We calculate the channel for each value of CLK, giving us the full ~24 hour hopping pattern.

At that point, we take observed packets, their timing gaps and look for where they fit in the pattern.

michaelp123, 2016-04-04:
Ok so if I understand the hopping sequence is determined once for all CLK values (i.e for each CLK value I will know the current channel) - right?

So what I need to know is the initial CLK value (i.e in t=0)?

dominicgs, 2016-04-04:
Correct, for each CLK value (given that we have the LAP and UAP) we can determine the channel.

CLK is a counter, it starts at 0 and continues to 2^27, what we're interested in is the current value of CLK.

Given that we now have a  mapping of channels to CLK values, and we receive packets on a given channel, we can narrow down the set of candidate values for CLK.  Over the course of multiple packets, we can narrow this down to a single value.  At this point, we know the master device's clock.

orgua, 2016-04-04:
Hi michael,
it is nice to see your interest in bluetooth, but all your questions have already been answered. you could read the paper dominic wrote years ago, the blog also adds valuable information. if you are really interested you could read the relevant ubertooth / libbtbb code. the latest code is well structured. this should be a platform for developers and questions regarding support and not for general comprehension. 
You are just scratching the surface and i am afraid that this will be a long question/answer-game if you are not eager to research yourself.

michaelp123, 2016-04-04:
OK it is much clearer now. Is it also pssible that different CLK values will generate the same channel index?

dominicgs, 2016-04-05:
@michaelp123 Yes, there are 2^27 timeslots and 79 channels, the channels repeat frequently, but the pattern is not repetitive over around 6 or more timeslots.

As @orgua mentioned, these issues are probably not the best place to discuss this, we have a mailing list or IRC channel that are more suited to discussions.  Both can be found [here](https://github.com/greatscottgadgets/ubertooth/wiki/Getting-Help#asking-questions).

I will leave this issue open for the time being to remind me to write something more formal (probably a blog post) about how this works, but that may be a while off.

michaelp123, 2016-04-07:
OK, thanks!
One more question - Is the clock always starting from 0 when we power up the device? Or is it initialized with a random value?

closed 2016-05-04 by dominicgs