greatscottgadgets/ubertooth #176: USB_claim_interface_error -6

raviwithu, 2016-04-05:
Hi Team,

   I am using Ubertooth-btle for sniffing BLE. Whenever btle finds CONNECT_REQ, the sniffing freezes and then when I run ubertooth-util -v I get "USB_claim_interface_error -6", finally disconnecting and connecting again it will work. This happening everytime and last packet that btle sniffes was always CONNECT_REQ. Could you guys guide me to resolve this

I am using Kali Linux 4.3.0-kali-amd64 in Vbox.
Ubertooth Firmware version - 2015-10-R1

Let me know if need any details

raviwithu, 2016-04-05:
Ran LIBUSB_DEBUG=8 ubertooth-dfu -d bluetooth_rxtx.dfu -r. I can see below

LIBUSB_DEBUG=8 ubertooth-dfu -d bluetooth_rxtx.dfu -r

raviwithu, 2016-04-05:
Ran below command and i can see Segmentation Fault
LIBUSB_DEBUG=8 ubertooth-dfu -d bluetooth_rxtx.dfu -r

ibusb: debug [libusb_exit] 
libusb: debug [libusb_exit] not destroying default context
libusb: debug [libusb_alloc_transfer] transfer 0x1781368
libusb: debug [libusb_submit_transfer] transfer 0x1781368
Segmentation fault

closed 2016-04-05 by raviwithu

thwillingham, 2016-10-15:
What did you do to resolve the error?  I am getting the same error.