greatscottgadgets/ubertooth #177: USB_claim_interface_error -6

raviwithu, 2016-04-05:
Hi Team,

I am using Ubertooth-btle for sniffing BLE. Whenever btle finds CONNECT_REQ, the sniffing freezes and then when I run ubertooth-util -v I get "USB_claim_interface_error -6", finally disconnecting and connecting again it will work. This happening everytime and last packet that btle sniffes was always CONNECT_REQ. Could you guys guide me to resolve this

I am using Kali Linux 4.3.0-kali-amd64 in Vbox.
Ubertooth Firmware version - 2015-10-R1

Ran below command and i can see Segmentation Fault
LIBUSB_DEBUG=8 ubertooth-dfu -d bluetooth_rxtx.dfu -r

ibusb: debug [libusb_exit]
libusb: debug [libusb_exit] not destroying default context
libusb: debug [libusb_alloc_transfer] transfer 0x1781368
libusb: debug [libusb_submit_transfer] transfer 0x1781368
Segmentation fault

Let me know if need any details

dominicgs, 2016-04-06:
USB connections to Virtualbox are often buggy and this may be causing your first problem.  Could you share the full ubertooth-btle command that you are running?

The second part of your issue, with ubertooth-dfu, is caused by Ubertooth changing USB ID when it enters DFU mode.  It can be solved by adding an additional USB filter to your virtualbox setup.  Run `ubertooth-util -f` to put the Ubertooth in to DFU mode, then take a look for a new USB device with the ID 1d50:6003 or ffff:0004 and add it to the list of devices to be passed through.

raviwithu, 2016-04-06:
am using ubertooth-btle -f -c BTsniff.pcap

raviwithu, 2016-04-06:
Also I tried ubertooth-util -f. I can see id 1d50:6003 and added to vbox list. But now all ubertooth commands are throwing "Could not open Ubertooth device"

dominicgs, 2016-04-07:
The Ubertooth commands will work with an Ubertooth in DFU mode, but the ubertooth-dfu command should now succeed.  Does it?

raviwithu, 2016-04-09:
I am still facing issue after changing to DFU mode 

libusb: debug [libusb_init] created default context
libusb: debug [libusb_init] libusb v1.0.20.11004
libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
libusb: debug [op_init] bulk continuation flag supported
libusb: debug [op_init] zero length packet flag supported
libusb: debug [op_init] sysfs can relate devices
libusb: debug [op_init] sysfs has complete descriptors
libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
libusb: debug [linux_get_device_address] scan usb2
libusb: debug [linux_get_device_address] bus=2 dev=1
libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
libusb: debug [linux_get_device_address] scan 2-1
libusb: debug [linux_get_device_address] bus=2 dev=2
libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
libusb: debug [linux_get_parent_info] Dev 0x1a28e70 (2-1) has parent 0x1a1fc80 (usb2) port 1
libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
libusb: debug [linux_get_device_address] scan 2-2
libusb: debug [linux_get_device_address] bus=2 dev=3
libusb: debug [linux_enumerate_device] busnum 2 devaddr 3 session_id 515
libusb: debug [linux_enumerate_device] allocating new device for 2/3 (session 515)
libusb: debug [linux_get_parent_info] Dev 0x1a2bd40 (2-2) has parent 0x1a1fc80 (usb2) port 2
libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
libusb: debug [linux_get_device_address] scan usb1
libusb: debug [linux_get_device_address] bus=1 dev=1
libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
libusb: debug [usbi_add_pollfd] add fd 7 events 1
libusb: debug [usbi_io_init] using timerfd for timeouts
libusb: debug [usbi_add_pollfd] add fd 9 events 1
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_init] reusing default context
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_open] open 2.3
libusb: debug [usbi_add_pollfd] add fd 10 events 4
libusb: debug [libusb_claim_interface] interface 0
libusb: debug [libusb_alloc_transfer] transfer 0x1a21878
libusb: debug [libusb_submit_transfer] transfer 0x1a21878
libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll fds modified, reallocating
libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
libusb: debug [linux_udev_event_thread_main] udev event thread entering.
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=2 status=-32 transferred=0
libusb: debug [handle_control_completion] handling completion status -32
libusb: debug [handle_control_completion] unsupported control request
libusb: debug [disarm_timerfd] 
libusb: debug [usbi_handle_transfer_completion] transfer 0x1a21878 has callback 0x7f2190e72a10
libusb: debug [sync_transfer_cb] actual_length=0
libusb: debug [libusb_free_transfer] transfer 0x1a21878
Switching to DFU mode...
libusb: debug [libusb_init] reusing default context
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_init] reusing default context
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_init] reusing default context
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_init] reusing default context
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_init] reusing default context
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_init] reusing default context
libusb: debug [libusb_get_device_list] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
Unable to find Ubertooth

dominicgs, 2016-04-12:
After switching to DFU mode, what does `dmesg` report?  How about `lsusb`?

dominicgs, 2016-05-04:
Are you able to provide any additional information on this issue?  Or was it fixed?

I will close this issue in 7 days if there is no new information.

mikeryan, 2020-12-03:
7 days 😏

closed 2020-12-03 by mikeryan