greatscottgadgets/ubertooth #181: Gerber files

darvin, 2016-04-19:
Build instruction says to send gerber files to PCB board maker. There is no instructions how to generate gerber files with KiCaD, or gerber files anywhere. Maybe gerber files for, say, oshpark will be nice to have in release?...

dominicgs, 2016-04-19:
The gerbers were part of the release when that was written but a number of users complained about being confused by the amount included in the release and the gerbers were one of the things removed.

Generating gerbers from KiCad is fairly straight forward, this tutorial should explain everything that you need: https://www.wayneandlayne.com/blog/2013/02/27/kicad-tutorial-gerber-file-generation/

darvin, 2016-04-20:
@dominicgs https://oshpark.com/shared_projects/MkaCMmcq - I shared the project on oshpark, may be useful to include in readme?..

darvin, 2016-04-20:
also added mouser project

https://github.com/greatscottgadgets/ubertooth/pull/182

closed 2017-03-01 by dominicgs