greatscottgadgets/ubertooth #183: Error cmTryCompileExec735061847

s6thsense, 2016-04-25:
Hi,

install of Ubertooth One fails on the following error: cmTryCompileExec735061847
I've tried to reinstall/update, but the result is the same.
This is the error log from CMake:
"Determining if the function btbb_pcap_create_file exists failed with the following output:
Change Dir: /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec735061847/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec735061847.dir/build.make CMakeFiles/cmTryCompileExec735061847.dir/build
make[1]: Entering directory '/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec735061847.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=btbb_pcap_create_file -I/usr/local/include    -o CMakeFiles/cmTryCompileExec735061847.dir/CheckFunctionExists.c.o   -c /usr/share$
Linking C executable cmTryCompileExec735061847
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec735061847.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=btbb_pcap_create_file    CMakeFiles/cmTryCompileExec735061847.dir/CheckFunctionExists.c.o  -o cmTryCompileExec735061847 -rdynamic$
CMakeFiles/cmTryCompileExec735061847.dir/CheckFunctionExists.c.o: In function `main':
CheckFunctionExists.c:(.text+0x14): undefined reference to `btbb_pcap_create_file'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec735061847.dir/build.make:89: recipe for target 'cmTryCompileExec735061847' failed
make[1]: Leaving directory '/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec735061847] Error 1
Makefile:118: recipe for target 'cmTryCompileExec735061847/fast' failed
make: *** [cmTryCompileExec735061847/fast] Error 2"

Kind regards,
s6thsense

dominicgs, 2016-04-25:
Which version of libBTBB do you have installed?

s6thsense, 2016-04-25:
Hi dominicgs,

I have version 2015-10-R1 installed

dominicgs, 2016-04-25:
Are you building on Windows?  I'm not sure how well that works, bu tyou will certainly want to run cmake with the `-DUSE_PCAP=OFF` argument.

s6thsense, 2016-04-25:
I'm trying to install Ubertooth One on a Rpi2 (forgot to mention)

dominicgs, 2016-04-25:
Could you list the steps you've taken to get to this stage?  Which commands or tools you have used?  Are you following the build guide for Debian/Ubuntu systems? (https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide)

s6thsense, 2016-04-25:
Yes, I'm following the build guide in the link you posted.
After command "make" it fails.

dominicgs, 2016-04-25:
The error message is something that should emerge when the cmake command is run.  Could you clear out your build directory and run the cmake command again, then paste the output here?

s6thsense, 2016-04-26:
Hi Dominicgs,

Here is the error log after cleaning up the build dir and run the cmake command again:

Determining if the function btbb_pcap_create_file exists failed with the following output:
Change Dir: /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec1916375310/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec1916375310.dir/build.make CMakeFiles/cmTryCompileExec1916375310.dir/build
make[1]: Entering directory '/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec1916375310.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=btbb_pcap_create_file -I/usr/local/include    -o CMakeFiles/cmTryCompileExec1916375310.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.0/Modules/CheckFunctionExists$
Linking C executable cmTryCompileExec1916375310
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec1916375310.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=btbb_pcap_create_file    CMakeFiles/cmTryCompileExec1916375310.dir/CheckFunctionExists.c.o  -o cmTryCompileExec1916375310 -rdynamic /usr/local/lib/libbtbb.so -Wl,-rpath$
CMakeFiles/cmTryCompileExec1916375310.dir/CheckFunctionExists.c.o: In function `main':
CheckFunctionExists.c:(.text+0x14): undefined reference to `btbb_pcap_create_file'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec1916375310.dir/build.make:89: recipe for target 'cmTryCompileExec1916375310' failed
make[1]: Leaving directory '/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec1916375310] Error 1
Makefile:118: recipe for target 'cmTryCompileExec1916375310/fast' failed
make: *** [cmTryCompileExec1916375310/fast] Error 2

Kind regards,
s6thsense

dominicgs, 2016-04-26:
Sorry, I don't think I was clear enough. Could you paste the output from
the cmake command, not make.

s6thsense, 2016-04-26:
I'm a newbie on this, so np.
This was the content off the "CMakeError.log" in the /ubertooth-2015-10-R1/host/build/CMakeFiles

dominicgs, 2016-04-26:
Could you please attach a copy of CMakeOutput.log too?  That will give us an idea of what succeeded and how we got to this state.

s6thsense, 2016-04-26:
here is the CMakeOutput.log:

The system is: Linux - 4.1.13-v7+ - armv7l
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc
Build flags:
Id flags:

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/3.0.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++
Build flags:
Id flags:

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/3.0.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec881564394/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec881564394.dir/build.make CMakeFiles/cmTryCompileExec881564394.dir/build
make[1]: Entering directory '/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec881564394.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec881564394.dir/testCCompiler.c.o   -c /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp/testCCompiler$
Linking C executable cmTryCompileExec881564394
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec881564394.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec881564394.dir/testCCompiler.c.o  -o cmTryCompileExec881564394 -rdynamic
make[1]: Leaving directory '/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec2158501722/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec2158501722.dir/build.make CMakeFiles/cmTryCompileExec2158501722.dir/build
make[1]: Entering directory '/home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/pi/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec2158501722.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec2158501722.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-3.0/Modules/CMakeCCompilerABI.c
Linking C executable cmTryCompileExec2158501722
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec2158501722.dir/link.txt --verbose=1
/usr/bin/cc     -v CMakeFiles/cmTryCompileExec2158501722.dir/CMakeCCompilerABI.c.o  -o cmTryCompileExec2158501722 -rdynamic
Using built-in specs.

s6thsense, 2016-05-03:
Hi,

Anyone have an idea?

Kind regards,
s6thsense

dominicgs, 2016-05-03:
Did you try adding `-DUSE_PCAP=OFF` to the cmake command?

i.e. `cmake -DUSE_PCAP=OFF ..` ?

s6thsense, 2016-05-04:
Hi Dominicgs,

I've tried the command you suggested.
The results are in both error logs:

[CMakeError.txt](https://github.com/greatscottgadgets/ubertooth/files/248576/CMakeError.txt)

[CMakeOutput.txt](https://github.com/greatscottgadgets/ubertooth/files/248587/CMakeOutput.txt)

s6thsense, 2016-07-04:
Hi Dominicgs,

Any  idea's?

Kind regards,
s6thsense

dominicgs, 2016-07-21:
I have no idea on this.  Looking back I don't seem to see the output that the `cmake ..` step prints to the terminal.  Do you think you could paste that here too?

s6thsense, 2016-07-27:
Hi Dominic

Thanks for you're response.
I ran out of options so I sold the Ubertooth.
But thanks for you're input anyway.

Kind regards,
6thsense

closed 2016-07-28 by dominicgs