greatscottgadgets/ubertooth #184: Ubertooth-scan

raviwithu, 2016-04-29:
Hi,

I mostly used ubertooth-btle but now i want to capture NFC, will ubertooth-scan captures that? i assuming ubertooth-btle capture only BLE not NFC. Also is there a way to capture in pcap file in ubertooth-scan like ubertooth-btle has?

mikeryan, 2016-04-29:
Ubertooth cannot capture NFC.

ubertooth-scan only works with BR devices: it unmasks undiscoverable devices and fetches their device name. It has no option to log to PCAP.

ubertooth-rx is similar in that it will discover undiscoverable devices. It does not do any further querying. It has flags for capturing to PCAP.

raviwithu, 2016-04-29:
Thanks a lot for the info. Also am looking for transimission of packets, i know Ubertooth supports it. but dont know where to start. Am looking for sending Vcard to PCE regitered client

closed 2020-12-03 by mikeryan