greatscottgadgets/ubertooth #185: Facing Problem with using ubertooth-util and also firmware upgradation

Nag5115, 2016-05-03:
Hii,
I am beginner to ubertooth project. After installation of all dependencies then i connected ubertooth one to vmware player. After that i'm trying to get the firmware version and also upgradation but it is showing error which is mentioned Below:

root@ubuntu:~# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 007: ID 1d50:6002 OpenMoko, Inc. 

root@ubuntu:/home/nag-anji# ubertooth-util -v
Segmentation fault (core dumped)

root@ubuntu:~# ubertooth-util -t
Starting TX test
Segmentation fault (core dumped)

root@ubuntu:/home/nag-anji/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-dfu -d bluetooth_rxtx.dfu -r
Segmentation fault (core dumped)

I seen and tried different possibilities. but i didn't get solution. so, please suggest solution to this issue.

Thanks...

dominicgs, 2016-05-03:
Which version of the Ubertooth tools are you using?  Which version of libbtbb did you install?

The USB ID for Ubertooth changes when we enter DFU update mode, so you'll need to add another USB ID to your VMware Player settings.  Try running `ubertooth-util -f`, which should set DFU mode, then looking at the list of available USB devices.  The Ubertooth ID will have changed and you should add another rule to pass this through to the guest OS.

Nag5115, 2016-05-03:
Thanks for replay Dominic Spill...
I installed the libbtbb-2015-10-R1 and also ubertooth-2015-10-R1. When i'm running the "ubertooth-util -f" also getting same error.

root@ubuntu:/home/nag-anji# ubertooth-util -f
ubertooth-util -f is no longer required - use ubertooth-dfu instead
Entering flash programming (DFU) mode
Segmentation fault (core dumped)
root@ubuntu:/home/nag-anji# ubertooth-util -dfu
Segmentation fault (core dumped)

root@ubuntu:/home/nag-anji# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 007: ID 1d50:6002 OpenMoko, Inc. 

Please suggest some more solutions to this issue.

Thanks

dominicgs, 2016-05-03:
You can connect pins 1 & 3 on the expansion header (as shown [here](https://github.com/greatscottgadgets/ubertooth/wiki/Ubertooth-One#pins-and-leds)) as you plug the Ubertooth in to enter DFU mode.  Unfortunately this has a rather short timeout, so you'll need to be quick to find the ID in the VMware list.

I'm not sure why you see those segmentation faults, I've not seen so many tools segfault so quickly before.  Do you have gdb installed?  Could you try finding where the segfault is coming from?

We generally recommend running on the host rather than in a VM due to USB problems, but if it's required, it should still work.

Nag5115, 2016-05-03:
Thanks for replay...
I connected the pins 1 & 3 on the expansion header as you mentioned. At that time it is entering into the DFU mode and all LED'S are blinking and the output of "lsusb" is shown below.

root@ubuntu:/home/nag-anji# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 012: ID 1d50:6003 OpenMoko, Inc.     ------------ ID In DFU mode.

Then what are the further proceedings to eliminate those errors. so, please provide information about this issue.

Thanks...

dominicgs, 2016-05-03:
Once you have that USB ID showing up, you can attempt to re-run the ubertooth-dfu command from earlier
`root@ubuntu:/home/nag-anji/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-dfu -d bluetooth_rxtx.dfu -r`

Nag5115, 2016-05-03:
Thanks for replay...
After it is entering into the DFU mode then i executed command you mentioned. The outcome of this i mentioned below. 

root@ubuntu:/home/nag-anji/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# ubertooth-dfu -d bluetooth_rxtx.dfu -r
Checking firmware signature
.......................................................................................................................................
Detached

what is the consequence of this output and what else i need to do??? so,please provide information about this issue.

Thanks...

dominicgs, 2016-05-03:
That's a successful firmware update.  Could you now try `ubertooth-util -v` ?

Nag5115, 2016-05-03:
Thanks...
No change i am getting same error which is  previously discussed. I am mentioning same below

root@ubuntu:~# ubertooth-util -v
Segmentation fault (core dumped)
root@ubuntu:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 023: ID 1d50:6002 OpenMoko, Inc. 

Is there any other alternative for this issue which i need to try again??? please provide info about this issue.

dominicgs, 2016-05-03:
Do you have gdb installed?  Could you run the command with gdb?

`gdb ubertooth-util`

You will see a gdb prompt and you should type:
`run -v`

Once the application segfaults, type:
`bt`
and copy the output in to this issue.

Nag5115, 2016-05-03:
Thanks for replay...
I followed your steps as above mentioned order. I am sending the response. so,please find below and suggest solution to this problem.

root@ubuntu:~# gdb ubertooth-util  ===== 1st command
GNU gdb (Ubuntu/Linaro 7.4-2012.04-0ubuntu2.1) 7.4-2012.04
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-linux-gnu".
For bug reporting instructions, please see:
<http://bugs.launchpad.net/gdb-linaro/>...
Reading symbols from /usr/local/bin/ubertooth-util...(no debugging symbols found)...done.
(gdb) run -v   ===== 2nd command
Starting program: /usr/local/bin/ubertooth-util -v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/i386-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0xb7e067d5 in libusb_submit_transfer () from /lib/i386-linux-gnu/libusb-1.0.so.0
(gdb) bt   ===== 3rd command
#0  0xb7e067d5 in libusb_submit_transfer () from /lib/i386-linux-gnu/libusb-1.0.so.0
#1  0xb7e08388 in libusb_control_transfer () from /lib/i386-linux-gnu/libusb-1.0.so.0
#2  0xb7fc33bb in cmd_get_rev_num () from /usr/local/lib/libubertooth.so.0
#3  0x080495fa in main ()
(gdb) 
 

Thanks...

dominicgs, 2016-05-03:
This one is new to me, I'm not sure what would be causing libusb to segfault there.  I wonder if this has to do with VMware Player.

Nag5115, 2016-05-04:
Thanks Dominic...
I'm trying to rectify this error but I'm unable to do it. So,Please provide some more alternatives to this problem which we didn't tried.

Thanks...

dominicgs, 2016-05-04:
Have you tried running on the hardware rather than in a VM?  Perhaps boot from an Ubuntu 16.04 image?

Nag5115, 2016-05-04:
Thanks for replay...
finally we made it. I run it is on the vmware instant contiki host. it's working fine. May be my vmware ubuntu 12.04 is having memory segmentation problem. Thanks for support. Output response I mentioned below.

user@instant-contiki:~$ sudo ubertooth-util -v
Firmware revision: 2015-10-R1
user@instant-contiki:~$ sudo ubertooth-util -V
ubertooth 2015-10-R1 (dominicgs@hydrogen) Mon Oct 12 22:30:56 BST 2015

Thanks...

dominicgs, 2016-05-04:
That's great.  I'm glad we got it to work.

I'll close this issue as no further action is required.

closed 2016-05-04 by dominicgs