greatscottgadgets/ubertooth #186: ubertooth-rx.c

sevstels, 2016-05-03:
String 65: 	while ((opt=getopt(argc,argv,"hVi:l:u:U:d:e:r:sq:t:")) != EOF) {
Probably must be->   :s:q:

dominicgs, 2016-05-03:
-s is scan mode, it doesn't take any arguments.

sevstels, 2016-05-03:
I try use mode -q

dominicgs, 2016-05-03:
I'm not sure what issue you are trying to raise here.  The code in your initial comment is valid and represents the options available in ubertooth-rx.

What issue are you reporting?

sevstels, 2016-05-03:
I made and presently testing ubertooth-rx port for Windows.
Under XP this code do not work correctly, -q options missed.

dominicgs, 2016-05-03:
That code is valid for the use of getopt(), it works on other platforms.  Are you using getopt() on Windows?

Have you modified the code in any way?
Which compiler are you using?
How did you configure the code?
Is there any build log output?

sevstels, 2016-05-03:
Visual Studio 2010

dominicgs, 2016-05-03:
The -q option relies on libpcap, which doesn't exist on Windows hosts.  None of the pcap options do.

sevstels, 2016-05-03:
Need go home... I explain tomorrow.

sevstels, 2016-05-04:
Here test result output:

argc = 7
argv[1]= -l
argv[2]= 43fa51
argv[3]= -u
argv[4]= f9
argv[5]= -q
argv[6]= dump.pcap

systime=1462343140 ch=39 LAP=43fa51 err=0 clk100ns=35097883 clk1=5616 s=0 n=-81 snr=81
Correct CRC! CLK6 = 0x10 found after 1 total packets.

Calculating complete hopping sequence.
Hopping sequence calculated.
26747 initial CLK1-27 candidates
systime=1462343149 ch=39 LAP=43fa51 err=0 clk100ns=119898397 clk1=19184 s=-10 n=-81 snr=71
systime=1462343155 ch=39 LAP=43fa51 err=0 clk100ns=183100091 clk1=29296 s=-21 n=-81 snr=60
systime=1462343164 ch=39 LAP=43fa51 err=0 clk100ns=270296028 clk1=43247 s=0 n=-81 snr=81
systime=1462343164 ch=39 LAP=43fa51 err=1 clk100ns=271896288 clk1=43503 s=0 n=-81 snr=81
Correct CRC! CLK6 = 0x11 found after 4 total packets.

Found hopping sequence in cache.
26747 initial CLK1-27 candidates
systime=1462343169 ch=39 LAP=43fa51 err=0 clk100ns=327097101 clk1=52335 s=-10 n=-81 snr=71
systime=1462343188 ch=39 LAP=43fa51 err=1 clk100ns=516694468 clk1=82671 s=0 n=-81 snr=81
systime=1462343189 ch=39 LAP=43fa51 err=0 clk100ns=524696231 clk1=83951 s=-5 n=-81 snr=76
systime=1462343221 ch=39 LAP=43fa51 err=0 clk100ns=840694588 clk1=134511 s=0 n=-81 snr=81
systime=1462343236 ch=39 LAP=43fa51 err=0 clk100ns=995091072 clk1=159215 s=0 n=-81 snr=81

Acquired CLK1-27 = 0x0768400
got CLK1-27
clock offset = 15447074.
uh oh, full_usb_buf not emptied

Wiresharc can read dump.pcap, but time marker wrong, probably I have incorrect porting code.
1	0.000000			UNKNOWN	43	WTAP_ENCAP = 161
Frame 1: 43 bytes on wire (344 bits), 43 bytes captured (344 bits)
Data (43 bytes)

dominicgs, 2016-05-04:
The -s option tells the Ubertooth to sweep through channels rather than staying on channel 39.  This is useful in environments with noise where Bluetooth devices may avoid a given channel.

Given that the -q option is now working, I will close this issue.

Wireshark not recognising the output from ubertooth-rx is already being tracked as another bug.

closed 2016-05-04 by dominicgs