greatscottgadgets/ubertooth #194: Facing problem with capturing traffic of ble with wireshark

Nag5115, 2016-05-13:
Hi All,
I follow below procedure for capturing the ble traffic but it is not working.
I'm using
 ubertooth-2015-10-R1
 libbtbb-2015-10-R1
 wireshark 1.6.5

Run the command: mkfifo /tmp/pipe
Open Wireshark
Click Capture -> Options
Click "Manage Interfaces" button on the right side of the window
Click the "New" button
In the "Pipe" text box, type "/tmp/pipe"
Click Save, then click Close
Click "Start"

In a terminal, run ubertooth-btle:
ubertooth-btle -f -c /tmp/pipe

after execution of the above command I'm getting below error in wireshark traffic.

I got the error """User encapsulation not handled: DLT=147, check your Preferences->Protocols->DLT_USER the steps you want are:"""

Click Edit -> Preferences
Click Protocols -> DLT_USER
Click Edit (Encapsulations Table)
Click New
Under DLT, select "User 0 (DLT=147)" (adjust this selection as appropriate if the error message -showed a different DLT number than 147)
---->>>Under Payload Protocol, enter: btle

after entering above step I'm getting below error.
"""error in column 'payload protocol': dissector not found"""

I'm new to this ubertooth project. so,please provide information about this issue.

Thanks...

dominicgs, 2016-05-13:
The BLE packet dissector is included with Wireshark 1.12 and later versions.

Wireshark 1.6.5 is over four years old and predates the BLE implementation in Ubertooth.  The BLE packet dissector will not build for version 1.6.5.

Nag5115, 2016-05-18:
Thanks for the replay,
I upgraded wireshark version to stable version of 2.0.3. its working fine.

Thanks for the support...

EMCP, 2016-05-20:
I'm not sure how to get started with the new wireshark.. it's saying "This version of wireshark does not save pipe settings" in the manage interfaces... window

Nag5115, 2016-05-20:
@EMCP yes,it is not showing the pipe settings but in offline mode we can filter the btle packets. I'm also facing same problem with newer version of wireshark with "mkfifo /tmp/pipe" settings... Is anyone try this with latest version of wireshark???

humberthumbert, 2016-06-24:
@EMCP @Nag5115 I faced the same problem that "This version of wireshark does not save pipe settings". And I think I have figured it out. There is an article on wireshark's wiki website taking about pipe. Here is the link [https://wiki.wireshark.org/CaptureSetup/Pipes](url)

So I tried the following steps:
$ mkfifo /tmp/sharkfin
$ wireshark -k -i /tmp/sharkfin &
$ ubertooth-btle -f -c /tmp/sharkin &

And it works for me now.

pekkanikander, 2016-08-25:
@humberthumbert Thanks for your instructions.  They work for me, too.

IgorGanapolsky, 2020-09-07:
I am facing the same error with Wireshark **3.2.6**

mikeryan, 2020-12-05:
@IgorGanapolsky can you please confirm you're using 2018-12-R1 along with Wireshark 3.2.6?

mikeryan, 2020-12-12:
Closing this due to inactivity. Please reopen if you're still having this problem.

closed 2020-12-12 by mikeryan