greatscottgadgets/ubertooth #195: error on compiling wireshark plugins, kali linux ARM

EMCP, 2016-05-16:
Was following along with the instructions to install the BTBB, but got stuck..

Raspberry Pi 2 running latest Kali Linux Release

```
root@kali:~/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/build# make
[ 20%] Generating plugin.c
Scanning dependencies of target btbb
[ 40%] Building C object CMakeFiles/btbb.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
In file included from /usr/include/wireshark/epan/packet.h:26:0,
                 from /root/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:32:
/root/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_payload_header1’:
/root/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:158:19: warning: implicit declaration of function ‘tvb_length_remaining’ [-Wimplicit-function-declaration]
  DISSECTOR_ASSERT(tvb_length_remaining(tvb, offset) >= 1);
                   ^
/usr/include/wireshark/epan/proto.h:165:13: note: in definition of macro ‘DISSECTOR_ASSERT’
   ((void) ((expression) ? (void)0 : \
             ^
/root/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_dm1’:
/root/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:259:9: error: ‘packet_info {aka struct _packet_info}’ has no member named ‘private_data’
    pinfo->private_data = &fake_acl_data;
         ^
/root/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_btbb’:
/root/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:287:6: warning: implicit declaration of function ‘tvb_length’ [-Wimplicit-function-declaration]
  if (tvb_length(tvb) > 0 && tvb_length(tvb) < 9)
      ^
CMakeFiles/btbb.dir/build.make:93: recipe for target 'CMakeFiles/btbb.dir/packet-btbb.c.o' failed
make[2]: *** [CMakeFiles/btbb.dir/packet-btbb.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/btbb.dir/all' failed
make[1]: *** [CMakeFiles/btbb.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
root@kali:~/Tools/libbtbb-2015-10-R1/wireshark/plugins/btbb/build# 

```

dominicgs, 2016-05-16:
Which version of Wireshark are you using?

EMCP, 2016-05-16:
```
root@kali:~/Tools/ubertooth-2015-10-R1/ubertooth-one-firmware-bin# wireshark -v
Wireshark 2.0.3

Copyright 1998-2016 Gerald Combs <gerald@wireshark.org> and contributors.
License GPLv2+: GNU GPL version 2 or later <http://www.gnu.org/licenses/old-licenses/gpl-2.0.html>
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiled (32-bit) with Qt 5.5.1, with libpcap, with POSIX capabilities (Linux),
with libnl 3, with libz 1.2.8, with GLib 2.48.0, with SMI 0.4.8, with c-ares
1.11.0, with Lua 5.2, with GnuTLS 3.4.11, with Gcrypt 1.7.0-beta, with MIT
Kerberos, with GeoIP, with QtMultimedia, without AirPcap.

Running on Linux 4.1.19-v7, with locale C, with libpcap version 1.7.4, with libz
1.2.8, with GnuTLS 3.4.11, with Gcrypt 1.7.0-beta.

Built using gcc 5.3.1 20160409.

```

dominicgs, 2016-05-16:
Unfortunately the interface to the plugins changed with Wireshark 2.0.  We have plugins in the git repository that support Wireshark 2.0+ but they have not been released yet.

The plugins do not rely on the rest of libbtbb, so they can be installed separately (don't install the git version of libbtbb as you will need to change the version of ubertooth that you have installed).

You can find the updated plugins here: https://github.com/greatscottgadgets/libbtbb/tree/master/wireshark/plugins

EMCP, 2016-05-17:
check out the new stuff, seems to work okay so far compiling.  I did noticed I still have 1 red LED on my board, thought that it would go green once I get everything installed.. will keep looking into it.  

Thanks for the quick response!

closed 2016-05-17 by EMCP

dominicgs, 2016-05-17:
@EMCP The LEDs are fixed colours, they are different colours to make it easier to tell them apart at a glance.  The red LED near to the USB connector indicates USB status.

EMCP, 2016-05-17:
I read that about 20 minutes after I posted .. heh

now I'm trying to get Kismet compiled on Kali.. seems `libnl-3-dev` isn't getting recognized on the `configure` script.