greatscottgadgets/ubertooth #196: Facing problem while compiling the btle wireshark plugin

Nag5115, 2016-05-18:
Hi All,
I'm at starting stage in this ubertooth project. while I'm compiling btle wireshark plugin in libbtbb library getting below errors. This error is same while compiling btbb wireshark plugin also.

/libbtbb-2015-10-R1/wireshark/plugins/btle/packet-btle.c: In function ‘dissect_btle’:
/libbtbb-2015-10-R1/wireshark/plugins/btle/packet-btle.c:408:11: error: ‘packet_info’ has no member named ‘private_data’
make[2]: *** [CMakeFiles/btle.dir/packet-btle.c.o] Error 1
make[1]: *** [CMakeFiles/btle.dir/all] Error 2
make: *** [all] Error 2

so,please provide solution to this problem.

Thanks...

dominicgs, 2016-05-19:
Which version of Wireshark?

Nag5115, 2016-05-20:
Thanks for replay...
I'm using wireshark stable version of 2.0.3 and I'm using all latest versions of libbtbb and ubertooth. while compiling the btle and btbb wireshark plugins it is showing this error "packet-btle.c:408:11: error: ‘packet_info’ has no member named ‘private_data" .

So, please provide solution to this problem...

Thanks...

EMCP, 2016-05-20:
@Nag5115 see https://github.com/greatscottgadgets/ubertooth/issues/195

dominicgs, 2016-05-24:
You do not need to build the BTLE plugins against any version of Wireshark after 1.12 as they are included in the Wireshark code.

closed 2016-05-24 by dominicgs