greatscottgadgets/ubertooth #197: Problem compiling wireshark Plugins

arunmagesh, 2016-05-19:
arun@arunmag:~/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/build$ sudo cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gn
u/wireshark/plugins/2.0.2 .. 
-- Plugin will be installed in: /usr/lib/x86_64-linux-gnu/wireshark/plugins/2.0.2
-- Configuring done
-- Generating done
-- Build files have been written to: /home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/build
arun@arunmag:~/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/build$ sudo make
[ 20%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
In file included from /usr/include/wireshark/epan/packet.h:26:0,
                 from /home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:32:
/home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_payload_header1’:
/home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:158:19: warning: implicit declaration of function ‘tvb_length_remaining’ [-Wimplicit-function-declaration]
  DISSECTOR_ASSERT(tvb_length_remaining(tvb, offset) >= 1);
                   ^
/usr/include/wireshark/epan/proto.h:165:13: note: in definition of macro ‘DISSECTOR_ASSERT’
   ((void) ((expression) ? (void)0 : \
             ^
/home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_dm1’:
/home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:259:9: error: ‘packet_info {aka struct _packet_info}’ has no member named ‘private_data’
    pinfo->private_data = &fake_acl_data;
         ^
/home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_btbb’:
/home/arun/Downloads/ubertooth/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:287:6: warning: implicit declaration of function ‘tvb_length’ [-Wimplicit-function-declaration]
  if (tvb_length(tvb) > 0 && tvb_length(tvb) < 9)
      ^
CMakeFiles/btbb.dir/build.make:93: recipe for target 'CMakeFiles/btbb.dir/packet-btbb.c.o' failed
make[2]: *** [CMakeFiles/btbb.dir/packet-btbb.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/btbb.dir/all' failed
make[1]: *** [CMakeFiles/btbb.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

This is the error i get. I have set the wireshark directory correctly, Still error. Any solutions?

dominicgs, 2016-05-19:
Which version of Wireshark are you using?  It looks like it is older that 1.10, which is the minimum version that we support.

arunmagesh, 2016-05-20:
I am using Wireshark 2.0.2. I suspect that the problem is in the plugin location(where *.so files are in)  in the cmake!
My version of wireshark has plugin at /usr/lib/x86_64-linux-gnu/wireshark/plugins/2.0.2/ 
But your version is at /usr/lib/x86_64-linux-gnu/wireshark3/plugins/

arunmagesh, 2016-05-20:
Issue solved 
:)

closed 2016-05-20 by arunmagesh

guqinchen, 2016-06-27:
@arunmagesh Hi, just encountered the same problem with you. I am using Ubuntu 16.04 LTS and the default wireshark installed is 2.0.2. How did you solve the error generated during btbb plugin compilation exactly? Thank you!