greatscottgadgets/ubertooth #198: Monitoring Date and Time

hectorbm94, 2016-05-24:
Hi,

I would like to know if it's possible to save into a file information of devices capture by ubertooth, such as the date when it was capture, the time it has been watched... so that I could create a web application with graphs where display all this info.

I have been trying with kismet, but I don't succeed. Thank you!

dominicgs, 2016-05-24:
We don't currently have the ability to log the printed information to a file, although you could try to log to pcap file.

Kismet will log to an older pcap format that we plan to retire, but it is currently supported by the Wireshark plugins.

An alternative approach would be to look at the current development branch of Kismet (git: nextgen-fields) which will have some web application support built in and sounds similar to what you are planning to do.  It does not currently support Ubertooth, but you may be able to write the code needed to add Ubertooth support.

If you are an IRC user, try asking about it in #kismet on freenode.

hectorbm94, 2016-05-25:
Okey, thank you Dominic, you solve my question!

closed 2016-05-25 by dominicgs