greatscottgadgets/ubertooth #201: Unable to sniff the pairing request of the mobile devices having BT version of 4.0

Nag5115, 2016-05-30:
Hi All,
I'm using the moto x play and samsung galaxy grand prime to make connection and also observe the different requests like pairing, connection for analyse the BLE. Both mobiles are having the Bluetooth version 4.0 . And also I'm using the below command.

" Sudo ubetooth-btle -c packet.pcap -f "

The above command is working between smart phone and CC2650 Sensor tag. But it is not working for connection between two smart phones. So,please provide some solutions to this problem.

Thanks...

dominicgs, 2016-05-31:
Are you sure that the two devices are communicating using BLE?  How are you creating the connection between them?  From Android settings or from an app?  If it's from the Android settings menu, it's much more likely that they are using regular Bluetooth to connect.  Ubertooth is not able to sniff the pairing between devices operating in the Bluetooth basic rate or enhanced data rate modes.

Nag5115, 2016-05-31:
Thanks for reply dominic spill,
Samsung Device:-Bluetooth v4.0, A2DP.
Moto X Device:-Bluetooth	 v 4.0

I'm connecting via the Android settings of bluetooth only. Then how can I connect these mobiles with BLE via an APP. Is there any free apps are available to connect then it is possible to sniff the different requests with ubertooth? so,please suggest some apps with sniffing possible.

Thanks...

dominicgs, 2016-06-01:
I do not know if it is possible to connect them through BLE.

Nag5115, 2016-06-01:
Thanks For Reply Dominic Spill,

Then is there any possibility to sniff these mobiles(BT basic rate) initial connection requests with Kismet and possibility to decrypt with btcrack. And which type of BLE devices(please specify) I need to buy for full BLE Analysis starting from Pairing to Encryption with ubertooth sniffing capabilities other than mobiles. I am using TI CC2650 BLE sensor tag with mobile app connection But it is not sending any pairing request, it is following like this scan request and response, connection request and response, service discovery and sensor data notifications in captured traffic.

So,please provide information about this issues.

Thanks...

dominicgs, 2016-06-01:
It is not currently possible to sniff the basic rate pairing with Ubertooth.

I don't know much about BLE devices, but any device that only uses BLE (rather than BLE+BR) should connect using BLE.

There is no pairing request in BLE, the connection request that you are seeing is probably what you are looking for.

Nag5115, 2016-06-03:
Thanks for the reply Dominic Spill,
BLE security analysis expert Mike Ryan developed one tool for cracking the PIN of BLE devices, Then that tool requires at least one pairing request and response to crack, he mentioned like this clearly. May be If I'm wrong then, In BLE devices pairing request is not there with over the air facility or permanently no such request??? if it is there BLE devices are using any other method for pairing like OOB Method.
So,please provide solution to this Issue.

Thanks...

dominicgs, 2016-06-03:
What issue are you expecting me to solve?  I cannot force your devices to produce additional requests.

Nag5115, 2016-06-04:
Thanks for reply Dominic Spill
Nothing like that, I'm Just clarifying my doubts. Thanks for the support.

Thanks...

closed 2020-12-03 by mikeryan