greatscottgadgets/ubertooth #204: Scan both LE and classic Bluetooth at the same time

asad-lesani, 2016-06-03:
Hi,

I'm using Ubertooth module to capture MAC addresses of Both LE and Classic Bluetooth module. Is it possible to to at the same time using `uberooth-rx`?

What is the best command to capture the MAC addresses with your module? I would like to test it in traffic network to see if I can get the travel time between two points based on detection time of a unique MAC address.

Thanks.

dominicgs, 2016-06-03:
BLE and BR use different radio settings, so the Ubertooth is configured to search for one or the other at any given time.  It may be possible to configure the radio to sniff both, but would lead to reduced reliability for both types of sniffing, so we have chosen not to do this.

asad-lesani, 2016-06-03:
So, then if I want to scan both of them should I use two different modules?

I just text something to see if with `ubertooth-rx` I can also capture MAC of BLE, I'm transferring file from one Galaxy s3 device to another galaxy s3 device. while transferring file, the module can see the MAC of transmitter. Does it mean that with ubertooth-rx I could capture the BLE too?

dominicgs, 2016-06-03:
Either two modules or write a program to switch back and forth between the modes.

ubertooth-rx only captures Bluetooth Basic Rate.

dominicgs, 2016-06-03:
Those commands will work to capture addresses.

BLE does not use the device address in the connection, but you will find
connection addresses with ubertooth-btle.

ubertooth-rx will give you LAPs, which are not unique, but are sufficiently
rare that they will serve your purpose.

asad-lesani, 2016-06-03:
I know that the LAP is just a part of MAC that is device name and the vendor name is not in that. But I think it should be really unlikely to have two devices with the same id but different vendor in an area.

For the BLE, you said the connection address and not device address, does this connection address change? is it possible to see two connections between two pairs of devices with the same connection address?

dominicgs, 2016-06-03:
I agree, your chance of overlapping LAPs is minimal.

I don't know a lot about the BLE addressing scheme, but I believe that there are random addresses and fixed addresses.  They are big enought hat your chance of overlap is still small.

asad-lesani, 2016-06-03:
So, if two BLE devices are connected are this connection address fix during the connection or it will be randomly change over time?

dominicgs, 2016-06-03:
Given that there is no bug here, I am going to close the issue.  I am happy to continue the discussion on the Ubertooth mailing list, here: https://lists.sourceforge.net/lists/listinfo/ubertooth-general

closed 2016-06-03 by dominicgs

asad-lesani, 2016-06-03:
OK

Thanks for your help.