greatscottgadgets/ubertooth #206: Kismet not capturing packets

munnashk, 2016-06-10:
Hi I have a Ubertooth One and i have installed all the required software and tools by following the link https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide in Kali linux 2.0 Virtual Box. 

I have kismet tool and was able to see the packets captures in pcapbtbb file.

after a day when i have started kismet and added a source and loaded the plugin but was unable to see any captures happening.

when i ran ubertooth-rx tool, i was able to see all the packets which will end with "uh oh, full_buf not emptied" but kismet did not capture any packet and displayed as "No Networks Seen'

and kismet server console with be filled with "uh oh, full_buf not emptied" 

Also kismet is creating an empty pcapdump files.

please look into attached screen grabs.

the ubertooth hardware has 4 leds ON i.e. rst, usb, 1v8, usr.

[image]
[image]
[image]

gmcremona, 2016-10-07:
Same problem here! i tried multiple instalations on diferent machines(including VMs) and got the same result in all of them. Even i tried with diferent versions of kismet because i thought it was his fault, but didn't work either. I follow this guide: 
https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide 
and the third party software installation:
https://github.com/greatscottgadgets/ubertooth/wiki/Third-Party-Software 
Always on Ubuntu 15.04. I will try in kali linux now(the latest version) but i don't expect different results.
Is there a enviorment(OS, Ubertooth & libbtbb & kismet versions) that you recomend?

dominicgs, 2016-10-08:
I recommend Ubuntu as a host OS for Ubertooth, but it should be running natively.  VMs do horrible things to our USB connection and this is very likely caused by using a VM.

gmcremona, 2016-10-08:
It worked! i used a VM with the latest kali linux version. Anyway, what solve the problem was the latest versions of libbtbb and ubertooth in the master banches. Also the latest release of kismet(2016-10-R1 i believe).

I think that the issue "uh oh, full_buf not emptied" doesn't allow you to gather packets when the kismet is running. Since this problem is solved in the master branch of ubertooth, it fixes kismet problem too

closed 2020-12-03 by mikeryan