greatscottgadgets/ubertooth #207: uh oh, full_usb_buf not emptied

asad-lesani, 2016-06-14:
Hi,

I have tried the module with another computer but still even for 15seconds scan I get "uh oh, full_usb_buf not emptied" error.
How can I solve it?

Thanks.

jbmontgomery4, 2016-06-30:
Can be solved using more than 2 buffers for usb transfers. You get the error above when both buffers are full. Will require changes to libubertooth. Wrote a modified libubertooth layer that uses a buffer list (10 buffers). Never see it have more than 2 buffers full and never get the error.

dominicgs, 2016-06-30:
Yes, that should solve the problem for most use cases, especially the ones that are temporary slow downs.

This should already be fixed by the ring buffer implementation that was added to libUbertooth in git.  We need to test it further before releasing.

asad-lesani, 2016-06-30:
Perfect guys,

When do you think you will release the new version?

dominicgs, 2016-07-01:
I don't have an ETA for the next release, but I would encourage you to try the code that is in the git repository to see if it fixes your problem.

If the problem persists, I'll need to @jbmontgomery4 for a look at the 10 buffer code.

asad-lesani, 2016-07-01:
Could you please give me some instructions for using that git?

elli89, 2016-07-05:
The current ringbuffer implementation stores USB packets and transfers them to the callback function. It is currently not used for the USB bulk transfer. But I like the idea to use it as a bigger receive buffer for bulk transfers in order to prevent buffer overflows.

I have written a patch and will file a PR. @asad-lesani could you please test it and tell us if that solves this issue?

mikeryan, 2020-12-12:
This should be fixed by #219. Please reopen if that's not the case.

closed 2020-12-12 by mikeryan