greatscottgadgets/ubertooth #213: cb_rx of ubertooth_callback.c

jbmontgomery4, 2016-06-28:
Not sure why this is getting the packet from the bottom instead of the top like the other cb's - 

Line 554: usb_pkt_rx* rx = ringbuffer_bottom_usb(ut->packets);

Also the r used in 

Line 679: 	if(infile == NULL && r < 0)
		    cmd_start_hopping(ut->devh, btbb_piconet_get_clk_offset(pn), 0);

is the result of the call in

Line 655: r = btbb_packet_get_type(pkt);

where the type of a packet is an uint8 which will never be negative. 

Is the expected value of r to be used for Line 679 instead meant to be the r returned from 

Line 654: r = btbb_process_packet(pkt, pn);

which returns a -1 if BTBB_CLK6_VALID and BTBB_CLK27_VALID with BTBB_FOLLOWING being set to 1. 

Thanks.

elli89, 2016-07-05:
Right, the result of btbb_process_packet() should be evaluated in order to start hopping. @dominicgs you moved the btbb_process_packet() line up in commit 7994326f287b583b59ca6202ff470faaae5745d0. What did you want to achieve with this change?

dominicgs, 2016-07-05:
@elli89 The changes in ubertooth_callback.c in that commit look like they were experimental and completely unrelated to the rest of the changes.  I should reverse that change.

mikeryan, 2020-12-05:
This was fixed in 993e4f3

closed 2020-12-05 by mikeryan