greatscottgadgets/ubertooth #214: cb_rx clock trimming

jbmontgomery4, 2016-06-30:
Not clear on the calibrate Ubertooth clock code found in the cb_rx function in ubertooth_callback.c. It seems for the most part that it prevents packets from being processed. Also the logic seems wrong. The trim_counter test is based on the clk_offset of the previous packet where the the cmd_trim_clock is set with the clk_offset of the current packet. See attached output file near the end. The trim_counter is incremented based on clk_offset > (CLK_TUNE_TIME + CLK_TUNE_OFFSET) and becomes > 8. However, the cmd_trim_clock is set to the clk_offset of the current packet  - CLK_TUNE_TIME which is a negative number. An explanation of what is being done here would be appreciated.

Attached is the modified ubertooth_callback.c file with added printfs, and the output of a run of the ubertooth-rx tool.

Thank you,
John Montgomery

[ubertooth-rx.txt](https://github.com/greatscottgadgets/ubertooth/files/342105/ubertooth-rx.txt)
[ubertooth_callback.c.txt](https://github.com/greatscottgadgets/ubertooth/files/342107/ubertooth_callback.c.txt)

elli89, 2016-07-05:
Your log shows a huge clock drift of the observed piconet. Between two received packets the piconet slips several bits away from the Ubertooth reference clock. The clock trimming procedure should correct that by slowing down or speeding up the Ubertooth clock for one cycle to match the piconet clock.

Since currently the host program receives USB packets from the Ubertooth stick in groups of eight packets, the trim procedure waits until it received eight packets that were out of sync before it tells the Ubertooth stick to trim its clock. Otherwise, in the worst case, it would trim the clock eight times during one USB transmission.

In your log the clock drifts too fast for the trimming algorithm to compensate it. I did not think of that case when I implemented it. Maybe in the future it would be a good idea to extend the clock drift correction in order to correct a constant clock drift and thus a linear clock deviation.

jbmontgomery4, 2016-08-22:
Just downloaded the code as of today. Still do not see any difference with the trimming of the clock. Still does not settle. Attached are execution captures showing what I still see. Any thoughts?

[ubertooth_rx_2.txt](https://github.com/greatscottgadgets/ubertooth/files/431017/ubertooth_rx_2.txt)
[ubertooth-rx.txt](https://github.com/greatscottgadgets/ubertooth/files/431014/ubertooth-rx.txt)

straithe, 2021-05-28:
@jbmontgomery4 are you still experiencing this issue?

closed 2021-06-28 by straithe