greatscottgadgets/ubertooth #215: Error building ubertooth on Mac with brew libbtbb

oxoocoffee, 2016-07-01:
I just installed libbtbb using home-brew. Here is info

16:20 $ brew info libbtbb
libbtbb: stable 2015-10-R1 (bottled), HEAD
Bluetooth baseband decoding library
https://github.com/greatscottgadgets/libbtbb
/usr/local/Cellar/libbtbb/2015-10-R1 (105 files, 866.9K) *
  Poured from bottle on 2016-07-01 at 15:55:39

Then I gig cloned https://github.com/greatscottgadgets/ubertooth

After adding to CMakeLists.txt

 if(NOT DEFINED CMAKE_MACOSX_RPATH)
   set(CMAKE_MACOSX_RPATH 0)
 endif()

cmake worked (with some complain about bluez which is linux only)

Now make complain about this

/home/rob/projects/GitHub/ubertooth/host/libubertooth/src/ubertooth.h:60:2: error: unknown type name 'btbb_pcap_handle'; did you mean 'btbb_pcapng_handle'?
        btbb_pcap_handle* h_pcap_bredr;
        ^~~~~~~~~~~~~~~~
        btbb_pcapng_handle

Looking at brew install, it only installed btbb.h to /usr/local/include (linked)

There is no definition of btbb_pcap_handle.

Looking at libbtbb source itself, btbb_pcap_handle is defined in pcap.c file.

I also installed libpcap with brew but there is no definition of btbb_pcap_handle

brew info libpcap
homebrew/dupes/libpcap: stable 1.7.4 (bottled), HEAD [keg-only]

dominicgs, 2016-07-02:
The libbtbb API has changed since the last release.  Ubertooth git head tracks libbtbb git head.  As you have libbtbb 2015-10-R1 installed, you will need to build the 2015-10-R1 release of Ubertooth.

Alternatively, you can remove the homebrew installed libbtbb and replace it with libbtbb git head.

dominicgs, 2016-07-06:
@oxoocoffee Did this fix your problem?

This formula should be added to homebrew in the future to prevent this from happening: https://github.com/Homebrew/homebrew-core/pull/2703

oxoocoffee, 2016-07-06:
Yes this worked like a charm :)   (10.11.6 Beta (15G24b))

11:37 $ clang --version
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

dominicgs, 2016-07-06:
That's great, thanks.  I'll close the issue.

closed 2016-07-06 by dominicgs