greatscottgadgets/ubertooth #222: Ubertooth Tools "Make"

fgranado, 2016-07-18:
Hey guys,

So following the build guide on Ubuntu 16.04, I got to ubertooth tools and can't make due to there not being a makefile. What could I do next? Thank you.

dominicgs, 2016-07-18:
Could you provide the output of the previous commands so that we can see what is happening?

fgranado, 2016-07-18:
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found LIBBTBB: /usr/local/lib/libbtbb.so  
-- Looking for btbb_pcap_create_file
-- Looking for btbb_pcap_create_file - not found
-- Checking for module 'libusb-1.0'
--   No package 'libusb-1.0' found
-- Could NOT find LIBUSB (missing:  LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR) 
-- Checking for module 'bluez'
--   No package 'bluez' found
-- Could NOT find LIBBLUETOOTH (missing:  LIBBLUETOOTH_LIBRARIES LIBBLUETOOTH_INCLUDE_DIR) 
-- Setting udev rule group to - plugdev
-- Ubertooth udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
   used as include directory in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
LIBUSB_LIBRARIES (ADVANCED)
    linked by target "ubertooth" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/libubertooth/src
    linked by target "ubertooth-btle" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
    linked by target "ubertooth-util" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
    linked by target "ubertooth-rx" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
    linked by target "ubertooth-ego" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
    linked by target "ubertooth-dfu" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
    linked by target "ubertooth-specan" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
    linked by target "ubertooth-debug" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src
    linked by target "ubertooth-dump" in directory /home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/ubertooth-tools/src

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/libbtbb-2015-10-R1/build/ubertooth-2015-10-R1/host/build/CMakeFiles/CMakeError.log".

Thank you by the way for the quick response. I've looked through your example, and I've changed the directories to:

 "~/ubertooth-2015-10-R1/host/build$ cmake .." 

but ended with same results. In the directories the files/folders that were created are:

CMakeCache.txt  CMakeFiles  cmake_uninstall.cmake  libubertooth  misc  python  ubertooth-tools

dominicgs, 2016-07-18:
You need to install the libusb header files. On Debian based systems this
is done by running `sudo apt-get install libusb-1.0-0-dev`

fgranado, 2016-07-18:
I'll try it out, thank you so much Dominic. I'm 34 minutes into your video https://www.youtube.com/watch?v=HU5qi7wimAM
Thank you for explaining alot on there. I'm new to bluetooth and trying to use it and develop this somehow (hopefully) for my senior project to graduate with my undergraduate degree. Do you mind if I ask you for help later on bluetooth related topics?

dominicgs, 2016-07-18:
If you would like to ask general questions about Bluetooth, I recommend our mailing list or our IRC channel.  Details of both can be found here: https://github.com/greatscottgadgets/ubertooth/wiki/Getting-Help

Please let me know whether or not this issue is solved for you and close the bug if it is.

Thanks.

fgranado, 2016-07-18:
Yep it worked sir, thank you so much!

closed 2016-07-18 by dominicgs