greatscottgadgets/ubertooth #223: Problem building firmware on Armbian (Debian) system

tomsaul, 2016-07-29:
When trying to build the firmware on my test system (OrangePi Plus 2E) with Debian/Jessie OS it fails.  I will attach a dump of the output to this report.  The system can build the libbtbb and tools without any problems - it is just the cross-compiler.

### Steps to reproduce
1. Clean install of OS and packages from Ubertooth build guide.
2. Grab master of source from GIT (download zip from web page, also tried actual git pull)
3. 'make' in the firmware directory

### Expected behaviour
All the associated builds should complete

### Actual behaviour
Fails linking blinky.  Direct attempts in individual directors also result in errors

### Version information
Debian Jessie

**Ubertooth firmware version (ubertooth-util -v):**
master

[make.txt](https://github.com/greatscottgadgets/ubertooth/files/391181/make.txt)

dominicgs, 2016-07-29:
It appears that you're missing `crt0.o` which I think may be part of newlib.  Could you try `apt-get install libnewlib-arm-none-eabi` ?  If that fixes it, I will update the instructions on the wiki.

tomsaul, 2016-08-01:
That did the trick - I went ahead and made the change to the Firmware 
build section on the wiki.

Tom Saul  --- (M) 360-620-9975 or (H) 360-297-4268

Web Site http://www.fatsheepfarm.com

dominicgs, 2016-08-01:
That's great.  Thanks for updating the wiki, I'll close this issue as fixed now.

closed 2016-08-01 by dominicgs