greatscottgadgets/ubertooth #224: ubertooth-rx LAPs do not match addresses in bluetoothctl scans

paulpatras, 2016-08-03:
### Steps to reproduce
1. Capture Bluetooth traffic for 30s and save LAPs: $ sudo ubertooth-rx | awk -F '[ =]' '{ print $6; fflush(stdout) }' >> laps.log
2. Get all unique LAPs: $ sort -u laps.log
(list of LAPs)
3. Run device discovery with BlueZ: 
$ blueetoothctl
[bluetooth]# scan on
(List of MAC addresses for devices in range)

### Expected behaviour
I would expect the LAPs to match the last 3 bytes of the MAC addresses identified with BlueZ. 

### Actual behaviour
Count matches, numbers do not.

### Version information
**Operating system**:  Ubuntu 16.04 (kernel  4.4.0-31-generic)

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 0.3 (2bc6502)

**libbtbb version:** libbtbb 0.4 (8d6c49c)

**Ubertooth firmware version (ubertooth-util -v):** git-2bc6502

paulpatras, 2016-08-08:
Adding more detailed logs:
$ bluetoothctl 
[NEW] Controller XX:XX:XX:52:AD:94 name [default]
[bluetooth]# scan on
Discovery started
[CHG] Controller XX:XX:XX:52:AD:94 Discovering: yes
[NEW] Device XX:XX:XX:25:2C:F1 Device Y

The only distinct LAP ubertooth-rx identifies is 9e8b33

Any suggestions would be much appreciated.

dominicgs, 2016-08-11:
The `9E:8B:33` LAP is the inquiry LAP, it's coming from your Bluetooth dongle scanning for the other devices (or from someone else scanning, it's universal).

The `91:DC:60` LAP has only been seen on two channels, so could be a false positive, or it could be a non-discoverable local device.

Ubertooth is able to passively reveal details about discoverable and non-discoverable devices that are within range, whereas bluetoothctl will actively probe for devices that are discoverable.  Seeing a discrepancy between the two is fairly normal.

paulpatras, 2016-08-11:
Thanks @dominicgs. All clear about the 9E:8B:33 LAP. 

The BT device I am using is discoverable, so I'd expect its lower part of the address to show up at some stage. Having run multiple scans for longer times (in what I believe is largely an environment with not many, if any other, devices), I still could not see the LAP of the target device. Instead, every time some other never the same values are reported. I enclose a list below. I'd be keen to know how you verified such LAPs do correspond to actual devices. 

^CSurvey Results
??:??:??:DC:0E:3E

^CSurvey Results
??:??:??:9E:8B:33
??:??:??:FC:22:63

^CSurvey Results
??:??:??:9E:8B:33
??:??:??:21:07:FE

^CSurvey Results
??:??:??:9E:8B:33
??:??:??:8C:8A:42

^CSurvey Results
??:??:??:9E:8B:33
??:??:??:4B:FB:2C

^CSurvey Results
??:??:??:9E:8B:33
??:??:??:4A:2E:36
??:??:??:F9:EC:BA

^CSurvey Results
??:??:??:9E:8B:33
??:??:??:FB:FF:FF
??:??:??:18:02:69
??:??:??:54:87:2B
??:??:??:A1:1D:31

paulpatras, 2016-08-18:
A quick update following more experiments. I increasingly believe the number of false positives is very high, and more so when scanning for BTLE devices with ubertooth-btle. My intuition is that often times packet boundaries are not identified correctly. I should probably look into the firmware code for preamble detection, but I am certain you are in a better position to do that, @dominicgs.

closed 2020-12-05 by mikeryan