greatscottgadgets/ubertooth #225: Ubertooth One not detected - OSX 10.11.6

newodahs, 2016-08-09:
Hello,
     I've been searching around and haven't found a definitive answer anywhere so thought I'd ask here:  Does the Ubertooth One work at all with OS X past 10.9?  I'm not having any luck despite following the current wiki instructions.

I posted my command output below (not the build commands; all of that appears fine).  Poking around the OS it doesn't even look like the Ubertooth One device is being detected by the OS.  This is a newly purchased device and the first time I've tried to use it, I'll try to pop it in somewhere else later too to verify all is well with it, but if this is a non-starter let me know.

### Steps to reproduce
1. Setup/Install Ubertooth on OS X per wiki
2. Plug in Ubertooth One
3. Run ubertooth-util -v

### Expected behaviour
Ubertooth One detected and ubertooth-util -v should return something...

### Actual behaviour
ubertooth-util -v
could not open Ubertooth device

### Version information
**Operating system**: Mac OS X 10.11.6

**Ubertooth tools version (ubertooth-rx -V):**  libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)

**libbtbb version:** listed above?

**Ubertooth firmware version (ubertooth-util -v):** Can't run...

mikeryan, 2016-08-09:
I'm one of the main developers on the project and I do most of my work on OS X, so it should be well supported. It sounds like there may be a hardware failure with your Ubertooth. Do any of the LEDs turn on? You should see two green LEDs at a minimum, and if ubertooth-util -v succeeds a red one will turn on too.

Check if it shows up under USB. Click the Apple menu, "About This Mac". In the window that opens click "System Report", and then USB. You should see "bluetooth_rxtx" in the list of devices.

newodahs, 2016-08-09:
I see one green and what appears to be a solid red, yellow/amber? and a very, very faint green

[image]

mikeryan, 2016-08-09:
That might be due to the firmware not being flashed properly. Let's try to get it into bootloader mode. Remove the Ubertooth from USB, then use a piece of wire or paperclip to jumper the two pins that are jumpered in the following picture:

https://pbs.twimg.com/profile_banners/19587312/1354079866/1500x500

Plug the Ubertooth back in with those pins jumpered. The three LEDs that are dimly lit in your picture should illuminate brightly in a pattern.

newodahs, 2016-08-09:
No dice, same pattern...

[Ubertooth Jumpered](https://goo.gl/photos/5chhYY6RDqRR2NeB7)

mikeryan, 2016-08-09:
Sounds like a hardware problem. Where did you purchase the Ubertooth from? Are you in the US?

newodahs, 2016-08-09:
I am in the US; I picked it up from the HakShop booth at Defcon just a few days ago.

mikeryan, 2016-08-09:
Checking with Dominic and Mike on next steps, but we should be able to get this replaced for you no problem. Hang tight.

newodahs, 2016-08-09:
Thank you, I appreciate the help!

mossmann, 2016-08-09:
newodahs: Please email me at mike@ossmann.com about getting a replacement.

closed 2016-08-09 by mossmann