greatscottgadgets/ubertooth #227: No Such File or Directory

fgranado, 2016-08-15:
Hey guys, so I'm installing your software on Raspbian and I went through the installation guide with no errors, but now that I have finished I connected the ubertooth, and tried to run 

`./ubertooth-scan`

and for the output I am getting.

`./ubertooth-scan: error while loading shared libraries: libubertooth.so.0: cannot open shared object file: No such file or directory`

I've looked online but haven't found a lot of information about this so I came here. If I go under /usr/local/bin/ all the files are there from your installation instructions, but still am missing something.

Also, side question, never thought about this till installing it now on the Pi, but does it matter what directory you run your build guide from? Or is there a specific directory I should be in to install prerequisites and libbtbb? Thank you guys.

dominicgs, 2016-08-15:
You need to run `sudo ldconfig` the first time that you install the library so that the OS picks it up.  A reboot will have the same effect.

As long as you follow the build guide you can start from any location.  The `build/` directories where you run `cmake ..` and `make` need to be in the appropriate places, as described in the build guide.

fgranado, 2016-08-15:
Hey Dominic,
Thank you for the quick response. So just to be on the safe side I went ahead and wiped out the Pi and reset it to default, updated it entirely, as well as the distro. I followed the directions carefully again, because I've tested this out on my laptop and it works fine, but again I ran into a problem. This time is installing Ubertooth tools.

When I run wget everything goes, but it does make an error regarding the certificate.
`ERROR: The certificate of 'github-cloud.s3.amazonawa.com' is not trusted.
ERROR: The certificate of 'github-cloud.s3.amazonawa.com' is not yet activated.
The certificate has not yet been activated`

This is followed by the tar command that brings up
`tar: This does not look like a tar archive
xy: (stdin): File format not recognized
tar: Child returned status 1
tar: Error is not recoverable: exiting now`
I've tried to adding the option J for the xy extension but no luck.

Installing libbtbb went fine, it's just now with ubertooth tools that the issue is coming. I'm not sure if this is related due to the Raspbian OS, I'm hoping you've encountered this in the past, if you haven't I can get you any information you need. Thank you again.

dominicgs, 2016-08-15:
This looks like a GitHub issue, perhaps a problem with their certificate.

The tar command is failing because the wget stage is failing, you're probably ending up with a text error message saved as ubertooth-2015.10.R1.tar.xz (or whatever you saved it as).

This isn't a bug in the Ubertooth process, simply a temporary problem with GitHub or your setup.  Could you try again?

fgranado, 2016-08-15:
I'm gonna go home and try it from my home network, I fear that maybe the network I am on is not trusting it. Once I get it up and running I'll let you know how it's going.

fgranado, 2016-08-17:
Hey Dominic, sorry I haven't responded. Yes the problem was that the local network I was using at the time actually was blocking GitHub. Once I got home I got it all running and working fine. I have two more questions but I'll make different tickets for those though so for anyone in the future looking for them will be able to find them and we can close this ticket.

dominicgs, 2016-08-17:
Thanks for getting back about that.  I'll close this issue.

If you have further questions you can also ask them on our mailing list or via IRC if you would prefer a discussion.  You can find details of both here: [Getting Help](https://github.com/greatscottgadgets/ubertooth/wiki/Getting-Help)

closed 2016-08-17 by dominicgs