greatscottgadgets/ubertooth #228: libUSB Error: Command Error:  (-9) Ubertooth running very old firmware found.

toby-f, 2016-08-18:
### Steps to reproduce
1. installed packages: cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev pkg-config libpcap-dev python-numpy python-pyside python-qt4 b
libusb-0.1-4 libusb-1.0-0-dev libusb-dev spectools
pip install --upgrade pyusb
2. Installed libbtbb and ubertooth from git 

But it doesn't work:

root@kali4bt:~# ubertooth-specan-ui 
Could not open Ubertooth device
('Failed to run: ', 'ubertooth-specan -d - -l 2400 -u 2483 -U -1')
root@kali4bt:~# ubertooth-btle 
libUSB Error: Command Error:  (-9)
Ubertooth running very old firmware found.
Please upgrade to latest released firmware.

**Operating system**: Kali Linux

**Ubertooth tools version (ubertooth-rx -V):** libubertooth 0.3 (579f25c), libbtbb 0.4 (8d6c49c)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware revision: 2015-10-R1

### Output
```
Insert any commandline or build output here
```
root@kali4bt:~# cat /etc/udev/rules.d/40-ubertooth.rules 
# Possible Ubertooth DFU
ATTRS{idVendor}=="ffff", ATTRS{idProduct}=="0004", MODE="0660", GROUP="plugdev"
# Ubertooth Zero
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6000", MODE="0660", GROUP="plugdev"
# Ubertooth Zero DFU
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6001", MODE="0660", GROUP="plugdev"
# Ubertooth One
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6002", MODE="0660", GROUP="plugdev"
# Ubertooth One DFU
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6003", MODE="0660", GROUP="plugdev"

dominicgs, 2016-08-18:
There's a new feature in the git host/firmware code that tries to check for matching interfaces betweent he two.  As the 2015.10.R1 firmware pre-dates the change, we make an assumption that it is "very old".  I think we need to improve the wording of the error here.

You can resolve this issue by building and updating the firmware from git.  It's pretty straight forward.  Instructions are available [here](https://github.com/greatscottgadgets/ubertooth/wiki/Firmware#developing-firmware)

toby-f, 2016-08-18:
that's it
thanks

closed 2016-08-18 by toby-f