greatscottgadgets/ubertooth #230: Ubertooth-blte -I (interfere command)

12424642, 2016-08-21:
Hello, 

I ran the command '**ubertooth-btle -f -I**' 
I was trying to see if it would interfere with the connection it was following. 

The test that i'm conducting involves a '**master vault bluetooth smart padlock**' and an **iphone6s**.  
I was expecting it to disrupt the connection in someway but it didn't seem to make any difference to the way it was sending requests to the device or the time it took for the device to respond. 

Am I completely off topic when I assume that the function of '**-I**' is to jam a connection or to introduce some kind of interruption to the connection resulting in packet loss?

davidHisense, 2018-10-17:
Hi, do you get what you want of the 'ubertooth-btle -f -I' command?
I also try to use the 'ubertooth-btle -f -I' command to interfere with connection between my two iphone6.And not yet find the jam of the connection:the two device communicated normally.

Could anyone give an example to test this function?

Thanks a lot.

mikeryan, 2020-12-05:
The `-I` option suffers from the same limitations as `-f`, namely that you have to win the race to see the `CONNECT_REQ` packet. If interference doesn't work, try try again.

Also, interference is currently broken as of 2018-12-R1.

closed 2020-12-05 by mikeryan