greatscottgadgets/ubertooth #231: Client plugin not showing as option in Kismet

digininja, 2016-08-24:
### Steps to reproduce
1. Link Ubertooth source into Kismet
2. Build Kismet
3. Install Kismet and plugins
4. Run Kismet
5. Kismet connects to the Ubertooth and collects packets but the UI plugin isn't shown in the console

### Expected behaviour
See the UI plugin

### Actual behaviour
Don't see the UI plugin

### Version information
Ubuntu 16.04

**Ubertooth tools version (ubertooth-rx -V):**

libubertooth 0.3 (6639145)

**libbtbb version:**
libbtbb 0.4 (8d6c49c)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware revision: git-6639145

Not sure if this is a Kismet problem or Ubertooth plugin problem so asking just in case.

The Kismet build works fine and the install works, I can see the files being placed in the right places but when I run Kismet the UI plugin isn't there. The server side plugin appears in the list so not sure what is happening.

I also tried building the plugin from its own directory and then installing it, again, looks like it runs OK and the files go to the right place but don't show up.

Kismet captures bluetooth packets so it is mostly working, just this last bit.

I've not done any modifications of the Kismet config file beyond adding the new log type and specifying the new source. 

Plugins are allowed to be loaded

```
 25 allowplugins=true
```

I've tried running Kismet as root and non-root, doesn't help.

Any ideas?

### Output
```
$ sudo KIS_SRC_DIR=/home/robin/tools/wifi/kismet-2016-07-R1/ make install
mkdir -p /usr/local//lib/kismet/
install -o "root" -g "root" -m 644 ubertooth.so /usr/local//lib/kismet/ubertooth.so
mkdir -p /usr/local//lib/kismet_client/
install -o "root" -g "root" -m 644 ubertooth_ui.so /usr/local//lib/kismet_client/ubertooth_ui.so

$ ls -l /usr/local//lib/kismet_client/
total 3908
-rw-r--r-- 1 root root 1600720 Aug 24 23:35 btscan_ui.so
-rw-r--r-- 1 root root  955704 Aug 24 23:35 spectools_ui.so
-rw-r--r-- 1 root root 1438488 Aug 24 23:49 ubertooth_ui.so

$ ls -l /usr/local//lib/kismet/
total 5980
-rw-r--r-- 1 root root  532864 Aug 24 23:35 alertsyslog.so
-rw-r--r-- 1 root root 1904920 Aug 24 23:35 btscan.so
-rw-r--r-- 1 root root  609280 Aug 24 23:35 kismet-syslog.so
-rw-r--r-- 1 root root 1320488 Aug 24 23:35 spectool_net.so
-rw-r--r-- 1 root root 1743608 Aug 24 23:49 ubertooth.so
```

[image]

toby-f, 2016-08-25:
Had the same Problem. After a clean reinstall the plugin runs.
Here my instructions:

Pakages:

```
sudo apt-get install libpcap0.8-dev libcap-dev pkg-config \
build-essential libnl-dev libncurses-dev libpcre3-dev \
libpcap-dev libcap-dev
```

Download Kismet http://www.kismetwireless.net/download.shtml

Symbolic Links:

```
ln -s ubertooth/host/kismet/plugin-ubertooth /opt/kismet/
ln -s ubertooth/host/kismet/plugin-ubertooth-phyneutral/ /opt/kismet/
```

Install

```
cd /opt/kismet/
sudo ./configure
sudo make && sudo make plugins
sudo make suidinstall
sudo make plugins-install
```

Compile and install the kismet plugin to enable kismet capture bluetooth packets

cd ubertooth/host/kismet/plugin-ubertooth
KIS_SRC_DIR=/opt/kismet make 
sudo KIS_SRC_DIR=/opt/kismet make install

Thats all. I needed three try... So try it again...

digininja, 2016-08-25:
I'll give it another try just in case.

Isn't the last set of make's redundant? By linking them in to the Kismet source and then making and installing plugins doesn't that do the same? I did try both but just checking.

dominicgs, 2016-09-18:
@digininja Which version of kismet are you running?  Which of the plugins did you build?

The phy-neutral plugin is designed for the git version of Kismet, whereas the regular plugin is designed for the latest Kismet release.

gmcremona, 2016-10-11:
Look into the folder "home/user/.kismet" the file "kismet_ui.conf". Search the variable "plugindir" and make sure that points to the place where your installation sent the plugins libraries. For example, in my case the variable pointed to "plugindir=/usr/lib/kismet_client" but the libraries went to "/usr /local/lib/kismet_client". Try to fix that path problem, it worked for me.

digininja, 2016-10-11:
Sorry, was on site when the first reply came though and forgot to look
again when I got home, I'll give it a test tonight.

anjanarajan, 2017-03-23:
Hi Sir,
Kismet server is working properly for me only thing i cant able to capture packet  using that plugin.No errors are coming.

mikeryan, 2020-12-05:
Legacy Kismet is no longer supported by this project.

closed 2020-12-05 by mikeryan