greatscottgadgets/ubertooth #232: Sniffer does not return to correct channel in ubertooth-btle -f

jagdap, 2016-08-27:
### Steps to reproduce
1. Run "ubertooth-btle -f -A39"
2. Observe BLE connection initiated on that channel
3. Wait for connection to end, sniffer will return to channel 37, not 39

### Expected behaviour
Sniffer running ubertooth-btle -f -A39 should follow a connection initiated on that channel and return to sniffing on that channel. I've observed the same behavior for channel 38 as well.

### Actual behaviour:
Sniffer returns to channel 37 (freq 2402) after connection complete instead of channel 39.

### Version information
Operating system:
Kali GNU/Linux Rolling

Ubertooth tools version (ubertooth-rx -V): 
libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)

Ubertooth firmware version (ubertooth-util -v): 
git-579f25c

### Possible fix
I've noticed that line 366 of bluetooth_rxtx.c contains the following:
`
channel = requested_channel;
requested_channel = 0;
`
Should the requested channel be reset to 0? I've compiled a version with the requested_channel line commented out and that seems to fix it, but I'm not sure what other effects this line of code may have or why it was put there in the first place.

dominicgs, 2016-08-27:
This looks like it's related to pull request #210

jagdap, 2016-08-28:
Yep - that's the same behavior I'm observing. I didn't see the pull request comments before.. sorry about that!

mikeryan, 2018-06-27:
This is fixed in the latest pre-release 2018-06-R1.

closed 2018-06-27 by mikeryan