greatscottgadgets/ubertooth #233: libUSB Error Timeout (-7)

Redndh, 2016-08-28:
Hi everyone,

Every time I want to use ubertooth-rx or ubertooth-follow, I get a _libUSB Error: Timeout: (-7)_
And following this, I cannot issue a _ubertooth-util -r_ as it ends up with the same error + _usb_claim_interface error -6_

I have followed many different setup tutorials. I believe that I have all the proper dependencies installed.
I am running Kali 2016.1 on bare metal. Running firmware 2015-10-R1 and I grabbed the required dependencies files from the ubertooth git.

I have tried and roamed around I can't find any solution so far.
Any advice?

Cheers
Louis

dominicgs, 2016-08-29:
How quickly do you get the error messages?
Do you get any expected output before the error message appears?

Redndh, 2016-08-29:
I get it right away, no output before it.

dominicgs, 2016-09-01:
Could you copy and paste the output from `lsusb -vd 1d50:6002` while the Ubertooth is connected to your system?

Also, please try `ubertooth-util -v` to log the firmware version (it looks like you may have done this already?)

The same goes for `ubertooth-rx -V` to show the host tools version.

It sounds to me as though your Ubertooth is responding to some USB communication but the bulk endpoints aren't working.  This could be a firmware issue or it could be hardware.

Redndh, 2016-09-02:
Here is the result for lsusb -vd 1d50:6002
```

Bus 004 Device 002: ID 1d50:6002 OpenMoko, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1d50 OpenMoko, Inc.
  idProduct          0x6002 
  bcdDevice            1.01
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 3 (error)
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              220mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x9940
  (Bus Powered)
  Debug Mode

```

Here is the output for ubertooth-util -v
Firmware revision: 2015-10-R1

And regarding the last command, the attribute -v is not recognized.

Also, I have been unable to capture any traffic since I had my Ubertooth. And I cannot flash the firmware... I followed the guide here: [](https://github.com/greatscottgadgets/ubertooth/wiki/Firmware)
I am getting another libusb error (-1)

I have tried from Kali 2016.1 bare metal & VM, Raspberry Pi Zero Kali, also a different machine running ubuntu. Same problem all across the board. I've tried using the apt-get install ubertooth and the install from the website... Same problems.

Thank for your help guys :)

dominicgs, 2016-09-02:
The errors for retrieving the strings are possibly permissions problems, but the `Device Status:     0x9940` is definitely not right.

The firmware update process often has some strange output that looks like an error, if you see the following (along with errors) it should be fine:
```
checking firmware signature
.....................................
Detached
```

I do not recommend updating firmware from a VM.  The firmware can be found in the `firmware-bin/` directory of [this archive](https://github.com/greatscottgadgets/ubertooth/releases/download/2015-10-R1/ubertooth-2015-10-R1.tar.xz)

Redndh, 2016-09-06:
Regarding the permissions, it's running on Kali Linux with full root rights, so permissions shouldn't be a problem theoretically.

I couldn't find any info about that device status error. Is there a reference guide regarding those?

I am not getting a firmware signature check unfortunately:
`ubertooth-dfu -r ubertooth-2015-10-R1/ubertooth-one-firmware-bin/bluetooth_rxtx.dfu
libUSB Error: Command Error:  (-1)
Switching to DFU mode...
Detached`

As I said I never managed to get anything out of the device, the only bit that works is the spectrum analyzer and that's it. I am really starting to think that the device could be faulty.

dominicgs, 2016-09-06:
Your `ubertooth-dfu` command has incorrect arguments there, it should be `ubertooth-dfu -d ubertooth-2015-10-R1/ubertooth-one-firmware-bin/bluetooth_rxtx.dfu`  The `-r` option causes the devices to reset in to normal firmware mode before checking the signature or performing the update.

Redndh, 2016-09-07:
My bad sorry about that, 

So it looks like the firmware is properly being flashed then?

`ubertooth-dfu -d ubertooth-2015-10-R1/ubertooth-one-firmware-bin/bluetooth_rxtx.dfu
libUSB Error: Command Error:  (-1)
Switching to DFU mode...
Checking firmware signature
.......................................................................................................................................`

dominicgs, 2016-09-07:
No problem, it looks like the firmware updated correctly.  Could you try running `lsusb -vd 1d50:6002` again?  Does the output contain the string descriptors?  What is the `Device Status:` field reporting?

Redndh, 2016-09-07:
Well it looks much better now :)

`lsusb -vd 1d50:6002

Bus 004 Device 002: ID 1d50:6002 OpenMoko, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1d50 OpenMoko, Inc.
  idProduct          0x6002 
  bcdDevice            1.01
  iManufacturer           1 http://github.com/greatscottgadgets/ubertooth
  iProduct                2 bluetooth_rxtx
  iSerial                 3 00000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              220mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
`

Does everything looks alright to you?

dominicgs, 2016-09-07:
That looks much better to me.  Could you try `ubertooth-rx` and try to generate some Bluetooth traffic?  (e.g. scan for devices from your phone)

Redndh, 2016-09-07:
Well, no usblib error in RX, still got it in FOLLOW.

Unfortunately still nothing, I've got a couple of Bluetooth speakers in pairing mode, a file transfer going from a laptop to a phone and another device scanning. I'm not getting anything :/

dominicgs, 2016-09-07:
What do you mean by "still got it in follow"?  You mean the `ubertooth-follow` tool?  That tool has other requirements, such as a connected Bluetooth dongle that it can access.

So you see none of the scans or the file transfer, but `ubertooth-specan-ui` works?

Redndh, 2016-09-07:
Sorry, I mean that I am still having the libusb error -7 while trying to to use the follow tool. 

And yes that's correct, the only things that works is the spectrum analyzer.

dominicgs, 2016-09-12:
I think we may need to investigate this further, could you email me with details of where you bought the Ubertooth

Redndh, 2016-09-13:
I had a friend of mine having a look over the week-end, I couldn't get it to work either and he's quite knowledgeable in terms of Linux and networking. We're thinking that the unit's got a hardware problem.
We're going to send it back for warranty!

Thanks a lot for your help!

closed 2016-09-13 by Redndh