greatscottgadgets/ubertooth #234: Potential bad memory access issue in BLE connection follower

jagdap, 2016-08-31:
### Steps to reproduce
1. Follow a BLE connection (ubertooth-btle -f)
2. Receive a 'short' (malformed) CONNECT_REQ packet
3. Sniffer freezes up and must be restarted

### Expected behavior
Sniffer should detect that the CONNECT_REQ is bad and it should ignore it

### Actual behavior
Sniffer freezes up, does not return to "listening" state

### Version information
**Operating system**:
Kali GNU/Linux Rolling

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)

**Ubertooth firmware version (ubertooth-util -v):**
git-579f25c

### Potential fix
It looks like CRC verification is disabled by default in the state machine within bluetooth_rxtx/connection_follow_cb(). I think this could cause a CONNECT_REQ (type 0x05) to come through with an invalid length (too short). This would mean tha ubertooth could read its 'connection interval' and 'channel increment' parameters from memory outside the scope of the packet variable.

As a temporary fix on my local firmware, I've added
```
crc_verify = 1; 
```
to the LINK_LISTENING state, and it (seems to have) eliminated the problem. However, I don't think that's the best fix since there is some value to catching malformed packets while listening. 

Am I missing anything with how this works? Does 'disabling' CRC verify allow malformed CONNECT_REQ packets to come through?

mikeryan, 2016-09-01:
Good catch. I think the right thing to do here is twofold:

1. Validate packet length and drop short packets
2. Validate connection parameters and ignore those that are out of spec

mikeryan, 2016-09-01:
Please see if the fixes I pushed to master help your issue.

dominicgs, 2016-09-18:
Hi @jagdap are you able to build and test the fixes that @mikeryan pushed to the git repository?  I'd like to release these fixes if you can confirm them.

jagdap, 2016-09-20:
So far, so good! I haven't come across the error again.  Thanks!

closed 2016-09-20 by jagdap