greatscottgadgets/ubertooth #235: Hi there,

mounish-p, 2016-09-16:
Hi there,

I'm trying to get the wireshark plugin to work in ubuntu 16.04. I have started following the guide 
When i run this command, ( ihave changed the location dir according to mine)
cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/plugins/2.0.2 ..

-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Plugin will be installed in: /usr/lib/x86_64-linux-gnu/wireshark/plugins/2.0.2
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'glib-2.0>=2.4.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
**-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found**
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/build
then 
make

**[ 20%] Generating plugin.c
Scanning dependencies of target btbb
[ 40%] Building C object CMakeFiles/btbb.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
In file included from /usr/include/wireshark/epan/packet.h:26:0,
                 from /home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:32:
/home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_payload_header1’:
/home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:158:19: warning: implicit declaration of function ‘tvb_length_remaining’ [-Wimplicit-function-declaration]
  DISSECTOR_ASSERT(tvb_length_remaining(tvb, offset) >= 1);
                   ^
/usr/include/wireshark/epan/proto.h:165:13: note: in definition of macro ‘DISSECTOR_ASSERT’
   ((void) ((expression) ? (void)0 : \
             ^
/home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_dm1’:
/home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:259:9: error: ‘packet_info {aka struct _packet_info}’ has no member named ‘private_data’
    pinfo->private_data = &fake_acl_data;
         ^
/home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_btbb’:
/home/mouni/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:287:6: warning: implicit declaration of function ‘tvb_length’ [-Wimplicit-function-declaration]
  if (tvb_length(tvb) > 0 && tvb_length(tvb) < 9)
      ^
CMakeFiles/btbb.dir/build.make:93: recipe for target 'CMakeFiles/btbb.dir/packet-btbb.c.o' failed
make[2]: *** [CMakeFiles/btbb.dir/packet-btbb.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/btbb.dir/all' failed
make[1]: *** [CMakeFiles/btbb.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2**

Is there something I can do to resolve this?

dominicgs, 2016-09-18:
This is a known issue with the 2.0+ Wireshark releases.  There are replacement plugins in git but they have not been released yet.

You can clone the git repository and build these plugins without updating the rest of your libbtbb installation.

mounish-p, 2016-09-19:
Thanks. i resolved it

closed 2016-09-19 by mounish-p