greatscottgadgets/ubertooth #236: libUSB Error: Command Error: (-9)

thwillingham, 2016-10-04:
### Steps to reproduce
1. download the latest ubertooth release
2. update the firmware
3. run ubertooth-rx

### Expected behaviour
The ubertooth-rx program should work

### Actual behaviour
I get an error:
libUSB Error: Command Error:  (-9)
Ubertooth running very old firmware found.
Please upgrade to latest released firmware.

### Version information
**Operating system**: Kali

**Ubertooth tools version (ubertooth-rx -V): libubertooth 0.3 (adeb80f), libbtbb 0.4 (8d6c49c)

**Ubertooth firmware version (ubertooth-util -v): Firmware revision: 2015-10-R1

thwillingham, 2016-10-04:
A reinstallation of the software seemed to fix the issue.

closed 2016-10-04 by thwillingham

dominicgs, 2016-10-04:
It looks like you previously had the git version of the Ubertooth tools installed, which introduces the `Ubertooth running very old firmware found.` error message.

Running the 2015 release and the matching firmware has solved the problem.