greatscottgadgets/ubertooth #241: ubertooth-rx returns data bytes as '00'

carvell, 2016-12-01:
In the current git master (f85a6a2), ubertooth-rx is returning all data bytes both to the screen and a pcap file as '00'.  There are the correct number of data bytes returned and the lap/uap etc. is detected fine, it's just the data bytes that are all '00'.

This behaviour doesn't exist with the 2015-10 release.  I can switch between 2015 release and git master and the behaviour is reliably broken in git master.

dand11, 2016-12-06:
I've narrowed this issue down to the following commit by @elli89 :
7a16e3a5922922aa3eb6cc86e68df7de037a7386

The data is printed correctly before the commit, and is shown as '00' after the commit.

elli89, 2016-12-06:
Uh oh. I will have a look at this issue tomorrow. Sorry for the inconvenience and thanks for your report.

dand11, 2016-12-12:
Any progress @elli89 ? Don't mean to pester but it'd be nice to know if you think it's an easy fix or not!

elli89, 2016-12-15:
The problem occurs whenever libbtbb detects a DM-3 or DM-5 packet. Those are multislot packets. Currently the Ubertooth firmware is not able to receive complete multislot packets when following a connection because it hops after each single slot. So I decided to store only the bitstream of single slots inside of the libubertooth fifo before passing it to libbtbb.

When libbtbb decodes a multislot packet but the bitstream is shorter than expected, the payload decode process fails and the decoded bluetooth packet contains only '00' data bytes.

Earlier implementations passed the bitstreams of the last 10 USB packets to libbtbb although they represent 10 individual bluetooth packets rather than one big packet. So it faked the missing payload bits by adding data from other bluetooth packets.

As a quick workaround libubertooth could also append zeroes to the received bitsteam before passing it to libbtbb. I will provide a patch for this workaround.

Both approaches generate "random" data bytes when unwhitening the faked payload bits without notifying the user which part of the payload is valid and which part is just random noise.

So what would be a good fix for this issue? I could imagine two ways that make sense:

1. if libbtbb does not get enough data bits it should tell the user "short packet, failed to decode payload"
2. it should unwhiten and unfec the provided data bits and tell the user "packet decoded but some bits are missing"

What do you think?
[Uploading zeropad-payload.patch…]()

dand11, 2016-12-18:
With the 'appending zeroes' approach, is it possible to determine which part of the unwhitened payload is real and which part is random noise?  If so, you could zero out the 'random noise' part and alert that user that because Ubertooth doesn't fully support multislot packets, not all bytes are present.

Even if the above was not possible, my preference would be to show the unwhitened bits, even if some were known to be random.  

I think it's really important to print something to the terminal informing the user that not all bytes are valid though.

dominicgs, 2017-03-01:
I've pulled across @elli89's work around for this issue, however as there is still a discussion about it here, I'll leave the issue open for the time being.

closed 2020-12-05 by mikeryan