greatscottgadgets/ubertooth #242: libUSB Error: Command Error:  (-9)

syncharny, 2016-12-02:
### Steps to reproduce
1. Compile libbtb from github
2. Compile ubertooth host from github
3. Compile firmware
4. Run command: _sudo ubertooth-dfu -d bluetooth_rxtx.dfu -r_ 

### Expected behaviour
Downloads firmware

### Actual behaviour
libUSB Error: Command Error:  (-9)
Switching to DFU mode...
Unable to find Ubertooth

Then the leds scroll up and down on the ubertooth. Ubertooth-util works fine, current version:
sudo ubertooth-util -V
ubertooth 2015-10-R1 (dominicgs@hydrogen) Mon Oct 12 22:30:57 BST 2015

### Version information
**Operating system**:
Ubuntu

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 0.3 (f85a6a2), libbtbb 0.4 (1eecca5)

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**
ubertooth 2015-10-R1 (dominicgs@hydrogen) Mon Oct 12 22:30:57 BST 2015

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
libUSB Error: Command Error:  (-9)
Switching to DFU mode...
Unable to find Ubertooth

dominicgs, 2016-12-02:
The error message `libUSB Error: Command Error:  (-9)` is sometimes misleading, especially when we're resetting the device.  The Ubertooth is resetting before returning from the USB request, so the host thinks that it has failed.  You can safely ignore this error in this one case.

However, the `Unable to find Ubertooth` is a problem, especially as the Ubertooth flashes the LEDs to indicate that it's in DFU mode.  For some reason the host cannot see the Ubertooth in DFU mode.

Are you running in a VM?

syncharny, 2016-12-02:
That's interesting, yes I am running a VM on my mac. I can try running from a native linux in the morning when I get to my computer. Thanks for the quick response, I'll let you know if it works or not.

dominicgs, 2016-12-02:
The change in USB IDs means that you'll need to add another USB address filter to your VM setup.  You may find you have performance issues with a VM, but some people report success.

You can also build most of the tools natively on a mac.

closed 2017-03-01 by dominicgs