greatscottgadgets/ubertooth #243: ('Failed to run: ', 'ubertooth-specan -d - -l2400 -u2483')

josch1, 2016-12-05:
I am trying to get the spectrum analysis working but the main issue seems to be that Ubuntu is not detecting the Ubertooth. The spectrum window pops up, then  I get this message.

sudo ubertooth-specan-ui
[sudo] password for josch1: 
Could not open Ubertooth device
**('Failed to run: ', 'ubertooth-specan -d - -l2400 -u2483'** 

### Version information
I'm running virtual machine 5.1.10 and ubuntu 16.10
Ubertooth-2015-10-R1.tar.xz
libbtbb-10-R1.tar.xz
Which tool I'm not for sure. I just followed the terminal commands from the build guide. 
I included a Photo of my device settings, also I just recently started to get this error message.
 
[image]
[image]

When the device is plugged in only the top two green lights are on, USB problem?
When my virtual machine is off all 3 light are on.

Its possible I could have installed something incorrectly, i'm new to Ubuntu, so I could have missed something. If someone finds a solutions please be specific lol, im still learning.

Thanks!

dominicgs, 2016-12-05:
It looks like the problem is that your host OS is claiming the Ubertooth rather than passing it through to the Ubuntu VM.  This means that the tools on the VM will not be able to find the Ubertooth.

If you run `lsusb` from the Ubuntu system there should be a device with the 1d50:6002 VID:PID combination (it will probably be listed as an Openmoko device).  If you don't see the device, then you will need to take a look at the settings on your host OS.  Is that Windows?

josch1, 2016-12-06:
Yes, my host is Windows 10 for my laptop. I am also running this on my Desktop which uses windows 7. I noticed on my desktop that lsusb shows my keyboard and mouse, but not the ubertooth device you mentioned.

Any ideas on what changes I should make to my OS?

dominicgs, 2016-12-06:
Do you see the same behaviour with both your laptop and desktop?  I think you need to check in device manager to see if anything else is claiming the Ubertooth before the VM can get to it.  I'm not sure what would be doing that though, there aren't any Windows drivers for Ubertooth that I know of.

josch1, 2016-12-06:
I'm using a surface pro 4, and under the ubuntu device settings my keyboard is seen as " Microsoft Surface Type Cover [2007]" but when I use lsusb it dose not show, probably because isn't used with a USB connection. I think im going to try a USB Drive and see if that works.

Il take a look at at my device manager, thanks!

josch1, 2016-12-07:
I managed to get my desktop to acknowledge the Ubertooth device and lsusb does show and open moko device  and the red LED is on. 

But I am still getting the same error with the spectrum analysts, any ideas on what this means?

dominicgs, 2016-12-07:
Could you run `ubertooth-util -v` from the VM on the desktop?

I think the cause here is probably that the VM is making USB communication unreliable.

josch1, 2016-12-11:
It ran and says firmware 2015-10.  

I got it working, I did some driver updates on my PC and reinstall the libbtbb file and I think that did it. 

Thanks alot !!

closed 2020-12-05 by mikeryan