greatscottgadgets/ubertooth #246: Kismet Server Failed

gracelee, 2017-01-10:
Hi I am trying to run kismet with ubertooth one, but I keep getting this error :( can anybody help me??

Kismet Server Failed
Locally started Kismet server exited unexpectedly with error0.

Then, says 

ERROR: Failed to open user plugin directory(/root/.kismet//plugins/): No such file or directory
INFO: Shutting down plugins
FATAL: Failed to load plugin 'aircrack-kismet.so' : This plugin was compiled for a different version of Kismet; Please recompile and reinstall it, or remove it entirely.

### Version information
Ubuntu 

**Ubertooth tools version (ubertooth-rx -V):**
ubertooth-2015-10-R1

**libbtbb version:**
libbtbb-2015-10-R1
**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

```

dominicgs, 2017-01-10:
Where did you install the Kismet plugins to?

It looks as though you're running Kismet as root, and it is therefore looking in `/root/.kismet/plugins/`, if that's not where they were installed then you will need to adjust the Kismet config to find them.

gracelee, 2017-01-10:
It's in /home/myname/.kismet
Should I move it to root directory ? 
Thank you!

gracelee, 2017-01-10:
I am following this tutorial  

http://cerescontrols.com/tutorials-3/sniffing-bluetooth-packets-with-kismet-and-wireshark-in-ubuntu-12-04/

gracelee, 2017-01-10:
Hi I changed the plugindir = /usr/lib/kismet_client to plugindir=/usr/local/lib/kismet_client in /home/grace/kismet_ui.conf. Then now run kismet with sudo kismet. Now it shows the error saying:

FATAL: Failed to load plugin 'aircrack-kismet.so' : This plugin was compiled for a different version of Kismet; Please recompile and reinstall it, or remove it entirely.
FATAL: Failure during activating plugins

INFO: Shutting down plugins

dominicgs, 2017-01-10:
That tutorial is roughly 4 years old.  I recommend our [Build Guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide) and the [Third Party Software](https://github.com/greatscottgadgets/ubertooth/wiki/Third-Party-Software) page for Kismet instructions.

gracelee, 2017-01-16:
Hi :) I followed the Build Guide, but building libbtbb doesn't work for cmake .. in build directory.

>cmake ..

    /home/libbtbb-2015-10-R1/build/lib/src/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  cmake/modules/GetGitRevisionDescription.cmake:77 (include)
  lib/src/CMakeLists.txt:28 (get_git_head_revision)

CMake Error at build/lib/src/CMakeFiles/git-data/grabRef.cmake:38 (string):
  string sub-command SUBSTRING requires four arguments.
Call Stack (most recent call first):
  cmake/modules/GetGitRevisionDescription.cmake:77 (include)
  lib/src/CMakeLists.txt:28 (get_git_head_revision)

-- Pcap include dirs set to /usr/include
-- Pcap library set to  /usr/lib/x86_64-linux-gnu/libpcap.so
-- Configuring incomplete, errors occurred!
See also "/home/grace/libbtbb-2015-10-R1/build/CMakeFiles/CMakeOutput.log".
See also "/home/grace/libbtbb-2015-10-R1/build/CMakeFiles/CMakeError.log".

relentless-warrior, 2017-07-07:
Hi,

I am getting the following cause for the Kismet server failure:

Kismet Server Failed
Locally started Kismet server exited unexpectedly with error0.

Then, says

ERROR: Failed to open user plugin directory(/home/username/.kismet//plugins/): No such file or directory
INFO: Shutting down plugins
FATAL: Failed to load plugin 'autowep-kismet.so' : This plugin was compiled for a different version of Kismet; Please recompile and reinstall it, or remove it entirely.

Could you please give a hint for it?

mikeryan, 2020-12-05:
Legacy Kismet is no longer supported by this project.

closed 2020-12-05 by mikeryan