greatscottgadgets/ubertooth #247: Ubertooth plugin visible but not loading in Kismet

Davedough, 2017-01-10:
Kali Linux ARM for Raspberry Pi 3b

### Steps to reproduce

1. Follow build guide to the letter
2. Edit kismet.conf in /usr/local/etc/ to add pcapbtbb to the list
3. Ensure the 40.ubertooth rules file is in the /etc/udev directory
4.  Add suid user to kismet group
5. Add Source ubertooth in Kismet successfully
6. Attempt to add plugin, find it in the list of server plugins, select it, close and then nothing...

### Expected behaviour
What should happen, and what I've been able to do on Raspbian with the same hardware was add the plugin to the source and it starts capturing data as expected. 

### Actual behaviour
When I Add the source 'ubertooth' after starting the server, and then select ` Plugins -> Select Plugins, I see the list of Server plugins and see "UBERTOOTH 2013-03-R Ubertooth plugin for Bluetooth baseba" so I highlight it, hit enter, hit Tab and then close. What I should see is then MAC addresses of bluetooth devices begin to appear, but that doesn't happen. I go and look back in the "Plugins" area and it shows "No Plugins Loaded..." No matter how many times I reboot, I can't get the plugin to load correctly.

### Version information
**Operating system**: Kali Linux for ARM

**Ubertooth tools version (ubertooth-rx -V):** 2015-10-R1 libubertooth 0.2 

**libbtbb version:** libbtbb 0.3

**Ubertooth firmware version (ubertooth-util -v):** 2015-10-R1

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here. Kismet v 2013-03-R1b

### Output
```
Insert any commandline or build output here
```

dominicgs, 2017-01-10:
If you run `kismet_server –c ubertooth` do you get output that suggests that it is sniffing Bluetooth addresses?

It sounds to me like the UI plugin is failing, but I'd like to be sure the backend part is working first.

Davedough, 2017-01-11:
Thanks for the quick reply. When I run that command, I get an output that says:

GOT PACKET ch=34 LAP=bfe020 err=0 clk100ns=125667
first sighting bfe020
GOT PACKET ch=6 LAP=bfe020 err=0 clk100ns=126263
new network bfe020
GOT PACKET ch=14 LAP=bfe020 err=0 clk100ns=127507
GOT PACKET ch=52 LAP=bfe020 err=0 clk100ns=128530
GOT PACKET ch=27 LAP=bfe020 err=0 clk100ns=132113
UAP = 0x2e found after 4 total packets
CLK = 0xe found after 1 total packets

And I ^C after that. It looks like it's collecting data. 

I should mention that I didn't install Wireshark as part of the instructions, I skipped and went straight to the Kismet installation.  After installing the Ubertooth tools and doing a sudo ldconfig, I moved onto the Third Party Instructions for Kismet. 

It's also worth mentioning that I have a GPS module connected to the Pi and it's collecting data and displaying fine through Kismet.

dominicgs, 2017-01-11:
That confirms that the main plugin is working and that we're looking at a UI plugin problem here.  I'll need to speak to people who know more about this than I do, I haven't used the Kismet UI plugin for Ubertooth.

No problem on the Wireshark install, it's not required or related to this, so you can skip it without problem.

rmelgares, 2017-06-19:
I'm have a similar issue. It appeared to work initially after having performed a fresh install of all the tools, but then stopped working after a while, and can't get it activate the UI plugin again, even though it's visible.

EMCP, 2018-01-02:
[image]

This is what I'm seeing with latest firmware and tools, on the suggested build of kismet

Seems I am getting the data out in the backend successfully though

LarryCheech, 2018-01-25:
I had the same issue, it seems you need to enable the ubertooth plugin through the kismet UI. 
[image]

mikeryan, 2020-12-05:
Legacy Kismet is no longer supported by this project.

closed 2020-12-05 by mikeryan