greatscottgadgets/ubertooth #248: cmake error

gracelee, 2017-01-16:
Hi Guys, I'm going though https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

However, I keep getting this cmake error.

Can you please help me fixting this error?

### Actual behaviour

myid@myid:~/libbtbb-2015-10-R1/build$ cmake ..
CMake Error at build/lib/src/CMakeFiles/git-data/grabRef.cmake:36 (file):
  file failed to open for reading (No such file or directory):

    /home/myid/libbtbb-2015-10-R1/build/lib/src/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  cmake/modules/GetGitRevisionDescription.cmake:77 (include)
  lib/src/CMakeLists.txt:28 (get_git_head_revision)

CMake Error at build/lib/src/CMakeFiles/git-data/grabRef.cmake:38 (string):
  string sub-command SUBSTRING requires four arguments.
Call Stack (most recent call first):
  cmake/modules/GetGitRevisionDescription.cmake:77 (include)
  lib/src/CMakeLists.txt:28 (get_git_head_revision)

-- Pcap include dirs set to /usr/include
-- Pcap library set to  /usr/lib/x86_64-linux-gnu/libpcap.so
-- Configuring incomplete, errors occurred!
See also "/home/grace/libbtbb-2015-10-R1/build/CMakeFiles/CMakeOutput.log".
See also "/home/grace/libbtbb-2015-10-R1/build/CMakeFiles/CMakeError.log".

### Version information
**Operating system**:
Ubuntu 16
**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version: 2015-10-R1**

**Ubertooth firmware version (ubertooth-util -v):**

```

dominicgs, 2017-05-04:
Which version of cmake was this with?  I think there may have been changes to cmake that we need to take in to account in the cmake configuration files.

anwarhamr, 2018-04-17:
I'm having the same problem.  I am doing the install today.  
cmake version 3.11.1

> CMake Error: The source directory "/Users/roblingstuyl/_sandbox/ubertooth-2017-03-R2" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

anwarhamr, 2018-04-17:
Found the problem.  The missing file is found in /host so after cd host/ && cmake .. && make it completed.

closed 2018-06-27 by mikeryan